m-labs/artiq #418: Can't access dataset_db variables in kernel

r-srinivas, 2016-04-29:
Using 1.0 rc3 on Windows, trying to access data_set variables in the kernel spits out an error.

For the following experiment,

```python
from artiq.experiment import *
import numpy as np

class database_test(EnvExperiment):
        
    def build(self):
        self.setattr_device("core")
        self.setattr_device("dds0")

    @kernel
    def run(self):
        self.set_dataset("dds_freq_bd1", 600, save=True, persist=True)
        print(self.get_dataset("dds_freq_bd1"))
```

I get the error message,

```
ERROR:worker(2008,database_test.py):root:Terminating with exception (RPCReturnVa
lueError: type mismatch: cannot serialize 600 as None (<function HasEnvironment.
get_dataset at 0x0000000004105BF8> has returned 600))
Core Device Traceback (most recent call last):
  File "C:\Anaconda3\artiq_test\repository\tests/database_test.py", line 20, in
__modinit__ (RA=+0x500)
    print(self.get_dataset("dds_freq_bd1"))
  File "<artiq>\coredevice\comm_generic.py", line 389, in _send_rpc_value
    lambda: "None")
artiq.coredevice.comm_generic.RPCReturnValueError(6): type mismatch: cannot seri
alize 600 as None (<function HasEnvironment.get_dataset at 0x0000000004105BF8> h
as returned 600)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-2016-04-21\lib\site-packages\artiq\master\worker
_impl.py", line 220, in main
    exp_inst.run()
  File "C:\Anaconda3\envs\artiq-2016-04-21\lib\site-packages\artiq\language\core
.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Anaconda3\envs\artiq-2016-04-21\lib\site-packages\artiq\coredevice\co
re.py", line 114, in run
    self.comm.serve(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-21\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 527, in serve
    self._serve_exception(object_map, symbolizer)
  File "C:\Anaconda3\envs\artiq-2016-04-21\lib\site-packages\artiq\coredevice\co
mm_generic.py", line 519, in _serve_exception
    raise python_exn
artiq.coredevice.comm_generic.RPCReturnValueError: type mismatch: cannot seriali
ze 600 as None (<function HasEnvironment.get_dataset at 0x0000000004105BF8> has
returned 600)
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 2008
```

Outside the kernel it's fine. It's not a totally necessary for us I think, but it could be nice to access those variables during the kernel if you want to switch the frequency of a laser between its cooling and detection frequencies during an experiment.

whitequark, 2016-04-30:
In short: the problem is that `get_dataset` does not specify a return type, and it defaults to `None`. So, when it returns something other than `None`, it's an error.

To work around this, wrap `get_dataset` like this:
``` python
def get_dataset_int32(self, name) -> TInt32:
     return self.get_dataset(name)
```

There is a section in the manual on this but it isn't up yet.

sbourdeauducq, 2016-04-30:
Or ``self.get_laser_frequency()``.

closed 2016-04-30 by sbourdeauducq