m-labs/artiq #419: Use of subclasses for experiments

r-srinivas, 2016-04-30:
Using 1.0 rc3 on windows I was looking at creating subclasses for experiments. It could be useful if you have similar experiments to avoid building everything again. I have two scripts in the same folder,

```python
from artiq.experiment import *

class Class(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("dds0")
    
    @kernel
    def run(self):
        self.sequence1()
        
    @kernel
    def sequence1(self):
        self.dds0.set(100*MHz)
```

and 

```python
from artiq.experiment import *
from experiment_class import Class

class Subclass(Class):
    
    @kernel
    def run(self):
        self.sequence1()
        delay(1*s)
        self.dds0.set(200*MHz)
```

In the GUI, I can see two entries `Class` and `Class1`, which both seem to correspond to `Class`, there's also an entry for  `Subclass`. All three entries seem to run fine. 

Aside from the additional entry in the GUI are there any issues with structuring experiments like this? Would it be better just to copy and paste the code from the class and add extra function instead of using subclasses?

sbourdeauducq, 2016-04-30:
Derive from ``HasEnvironment`` instead of ``EnvExperiment`` if you do not want it listed in the GUI. Then ``Subclass(Experiment, Class)`` to get ``Subclass`` listed.

sbourdeauducq, 2016-04-30:
And note that your parent class does not have to derive from anything, unless you want it to access datasets, device db, etc. For using ``@kernel`` all you need is a ``core`` attribute.

closed 2016-04-30 by sbourdeauducq

r-srinivas, 2016-05-06:
That worked, thanks. 

I'm thinking of having all the subcomponents we could use for various experiments in one specific python script and just import from there. For example,

in `repository\subcomponents\subcomponent_test.py' we could have 

```python
from artiq.experiment import *

class Subcomponent(HasEnvironment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")
    
    @kernel
    def cool(self):
        self.ttl2.pulse(20*us)
```

with an `__init__.py` file in the folder, then import those subcomponents to the various experiments. Does that seem sensible or is there a better way of doing this?

sbourdeauducq, 2016-05-07:
Sounds OK a priori. Note that you do not have to use inheritance but can also create several sub-experiments - see the arguments_demo example.