m-labs/artiq #420: Sorting and modifying entries of dataset panel on GUI

r-srinivas, 2016-04-30:
Based on how we were using HFGUI before, it seems that the dataset panel will have many persistent entries that correspond to various DDS frequencies, flopping frequencies, pi times, etc. It would be useful to be able to sort those out from the data from an experiment run. 

What I was thinking of was something along the lines of being able to click the `Persistent` header and it will list all the non-persistent ones first. This would just make it easier to distinguish the data that was just accumulated from the persistent data that's always there.

Additionally, it would be nice to able to modify values of these persistent data directly from the gui instead of having to write an experiment. Basically be able to modify the value of entries by right-clicking or something.

sbourdeauducq, 2016-05-01:
You can modify datasets with ``artiq_client set-dataset``. No need to write an experiment.

sbourdeauducq, 2016-05-01:
As for sorting, always showing persistent datasets first is easily doable. But considering the usual experience with Qt models/trees/lists, making sorting configurable at runtime is asking for trouble.

r-srinivas, 2016-05-01:
I think having the non persistent data like flopping brightness first makes
it easier to sort probably. If it's not configurable I think that's fine.
The main issue is that when it's sorted alphabetically it could make it
difficult to distinguish between the persistent data and what was recently
obtained from an experiment.

sbourdeauducq, 2016-05-02:
> The main issue is that when it's sorted alphabetically it could make it
> difficult to distinguish between the persistent data and what was recently
> obtained from an experiment.

Having the persistent data first won't solve this problem: if you run the experiment again, it will replace the existing datasets, not create new ones. And what about experiments that produce persistent data?

An easy solution is to log or print the dataset changes in the experiment.

r-srinivas, 2016-05-02:
Following flopping_f_simulation, I meant having the non-persistent data like `flopping_f_brightness`, `flopping_f_frequency` first before data that we want on every GUI startup like `F0`. We would probably have many more entries like `F0` rather than the non-persistent data that are constantly being replaced.

sbourdeauducq, 2016-05-02:
We can also filter by persistent/temporary, like we already filter by name. That's easier to implement than sorting, because for the latter, we need to handle the corner case of where to put ``foo`` in the tree when ``foo.bar`` is persistent and ``foo.quux`` is temporary.

jordens, 2016-05-02:
@r-srinivas in practice, if you properly organize your experiment and dataset names, you will have persistent datasets with different names, prefixes and in different groups than the transient ones. Isn't this only an issue if little thought went into assigning the names of your datasets?

Modification would need to be restricted to trivial datatypes. I remember having modification of datasets on some list of items but apparently it was dropped for a reason.

sbourdeauducq, 2016-05-02:
One way we could support complex modifications of datasets is by popping up a transient Python console with the dataset easily accessible in its global namespace.

jordens, 2016-05-02:
Yes. But also on a console/command line editing a (20,20) array is not all that practicable.

dleibrandt, 2016-05-02:
I agree that it would be useful to be able to edit datasets directly from the GUI.  I also think that it would be fine to simply restrict this capability to scalar datasets.

On a tangentially related note, it would be nice to have a GUI widget for 2D arrays.  We have something like this in the aluminizer which we use for setting trap voltages.

sbourdeauducq, 2016-06-18:
Dashboard now has basic dataset editing capabilities.

closed 2016-06-18 by sbourdeauducq