m-labs/artiq #421: Increasing current drive for nist_qc2 DDS bus signal integrity

r-srinivas, 2016-05-02:
We had some issues when loading the backplane with 11 DDS`. They started failing to programme starting from DDS0, which seems to be due to some bus signal integrity issues. This seems to be fixed by increasing the current drive from 12 to 24 mA. This enables us to handle a fully loaded bus. Could we apply the following patch to gateware/nist_qc2.py?

```diff
diff --git a/artiq/gateware/nist_qc2.py b/artiq/gateware/nist_qc2.py
index 12c775f..a7bd746 100644
--- a/artiq/gateware/nist_qc2.py
+++ b/artiq/gateware/nist_qc2.py
@@ -43,7 +43,7 @@ def get_fmc_adapter_io():
                 Subsignal("wr_n", FPins("FMC:LA24_P")),
                 Subsignal("rd_n", FPins("FMC:LA25_N")),
                 Subsignal("rst", FPins("FMC:LA25_P")),
-                IOStandard("LVTTL")),
+                IOStandard("LVTTL"),  Misc("DRIVE=24")),
 
             ("i2c_fmc", next(i2c_fmc),
                 Subsignal("scl", FPins("FMC:IIC_SCL")),
```

sbourdeauducq, 2016-05-03:
There are no particular issues. Did you get the backplane to work with 11 DDSes?

r-srinivas, 2016-05-03:
Yes, 11 seemed to work fine with flterm. The io_update signal at least looked better on the scope as well.

closed 2016-05-03 by sbourdeauducq

dhslichter, 2016-05-03:
Note that we only used 11 DDS cards because one of the slots on the backplane wasn't working correctly; we will test with a full load of 12 at a later stage.

r-srinivas, 2016-05-21:
Tested with 12 DDS boards with flterm and it seems to work.

commit 90e678a4 (2016-05-03): gateware/nist_qc2: increase DDS bus drive strength. Closes #421
commit 16fdebad (2016-05-03): gateware/nist_qc2: increase DDS bus drive strength. Closes #421