m-labs/artiq #423: Passing kernel function pointers

cjbe, 2016-05-04:
I am trying to write a kernel function that repeatedly calls another kernel function received as an argument - my code, below, fails to compile.
If I comment out the kernel decorator for funcSample() the code runs as expected, but is of course very slow.

Is this a compiler bug, or am I doing something silly?

```python
class FuncPointerTest2(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.sub = SubClass(*self.managers())

    @kernel
    def funcSample(self, func, N):
        results = 0
        for i in range(N):
            results += func()
        return results

    @kernel
    def func(self):
        return 1

    def run(self):
        results = self.funcSample(self.func, 10)
        print(results)
```

2016-05-04 17:58:21,286 ERROR:worker(21191,funcPointerSubmoduleTest.py):root:While compiling <repository>/tests/funcPointerSubmoduleTest.py
<repository>/tests/funcPointerSubmoduleTest.py:32:28-32:30: error: mandatory argument 'self' is not passed
            results += func()
                           ^^
<repository>/tests/funcPointerSubmoduleTest.py:32:24-32:28: note: the called function is of type (self:'a)->int(width='b) delay('c)
            results += func()
                       ^^^^
2016-05-04 17:58:21,288 ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 21191

whitequark, 2016-05-04:
It is a bug. The function should be converted to a method.

sbourdeauducq, 2016-05-07:
@cjbe Do you really need function pointers?

cjbe, 2016-05-07:
I think function pointers are necessary for some common experiment patterns, where one wants to pass an 'experiment' kernel function into some library machinery.

An example is a function that implements a one dimensional scan, where each scan point consists of N repeats of an experiment. It would be nice to package up all the boiler-plate for this scan into a library function which takes a vector of points and a kernel function pointer.

jordens, 2016-05-07:
Would mixin classes/inheritance not work just fine?

whitequark, 2016-05-08:
@jordens That doesn't currently work either.

jordens, 2016-05-08:
Sure. But is has a different and maybe easier implementation and it can make experiment code more readable.

whitequark, 2016-05-08:
that one is harder to implement, which is why we pushed it to 2.0

whitequark, 2016-05-08:
As a temporary workaround, the `run` function can be marked as `@kernel` (or the relevant parts extracted into one).

commit 4e5d7529 (2016-05-09): compiler: fix quoting of methods (fixes #423).
commit 9196cdc5 (2016-05-09): compiler: fix quoting of methods (fixes #423).
closed 2016-05-09 by whitequark