m-labs/artiq #425: Seamless handover between experiments

jordens, 2016-05-05:
* reduce hardware idle time
* reduce duty cycle jitter
* may need proper cleanup experiment that evicts rogue experiments and clears/resets FIFOs, time lines

see also #391

sbourdeauducq, 2016-05-06:
There is also the question of the idle kernel.

sbourdeauducq, 2016-05-06:
I propose that if there is no kernel executing, and the RTIO counter exceeds ``now``, then:

1. If there is a idle kernel programmed, RTIO reset + run the idle kernel.
2. RTIO reset when the next kernel is run.

jordens, 2016-05-06:
There is another thing: the idle kernel is potentially without any inputs. In that case the idle kernel will quickly fill the output fifos and advance the timeline quite a bit. That will delay the execution a higher priority experiment which is not desirable. To counteract that, people can `sync()` in the idle kernel. But that kills most of the slack.

Otherwise that behavior sounds ok. It is a bit hidden and implicit though. And the reset/clear machinery still needs to be exposed.

sbourdeauducq, 2016-05-06:
The "RTIO resets" I mentioned stop the kernel CPU, empty the FIFOs, and when the next kernel is executed set ``now`` to the RTIO counter value plus a margin.

jordens, 2016-05-06:
Yes. But if `now >> rtio_counter` and/or there is undesired stuff in the FIFOs the way to fix that is to write an experiment that resets `now` into the past, exits and through that very indirectly clears the FIFOs.

sbourdeauducq, 2016-05-06:
Setting ``now`` in the past and exiting is another error under inter-experiment seamless handover; it would cause an underflow in the next experiment anyway.

sbourdeauducq, 2016-05-06:
I would add an explicit reset.

sbourdeauducq, 2016-06-29:
Done. All resets and startup tests of time cursor vs. time counter are explicit and managed entirely by kernels (e.g. https://github.com/m-labs/artiq/blob/c8dc6ca07c12988ca27b56950911045101f476cf/artiq/examples/master/idle_kernel.py#L11-L14)

closed 2016-06-29 by sbourdeauducq