m-labs/artiq #426: new logo looks like a bicycle crank

dhslichter, 2016-05-05:
How about making the wrench open-ended instead of a closed circle?  :)

[image]

jordens, 2016-05-06:
[image]

jordens, 2016-05-06:
+1 to the *next* logo (3.0) actually being bicycle-themed.

dhslichter, 2016-05-06:
The wrench you show is indeed a wrench, but I will just say that every person in the lab I have showed the new logo to so far (approx 5 people) has first thought it was a bike crank.  Just FWIW.

Will the logo change with every major version?  @r-srinivas has pointed out that the current logos do not reflect the modern understanding of atomic structure ;)

jordens, 2016-05-06:
The perceptual bias towards recognizing bicycle parts in everything could be a simple Boulderism. And even if it *is* in fact a bicycle crank, would that be a bug or a feature?

I guess the logos can very well mirror the progression towards better tools. Ideas and comments welcome.

Can there ever be a correct visualization of (the modern understanding of) atomic structure? Or do we all know or suspect that this is impossible?

There are certainly cases where even a known-to-be-incomplete and maybe not even fully correct visualization of quantum mechanics leads to correct and even uncovers better results than would otherwise be obtained:
https://www.scienceathome.org/games/quantum-moves/game
http://www.nature.com/news/human-mind-excels-at-quantum-physics-computer-game-1.19725

sbourdeauducq, 2016-05-06:
> @r-srinivas has pointed out that the current logos do not reflect the modern understanding of atomic structure ;)

I'm well aware of this problem and had told the graphics designer. I had showed her this poster:
http://darksilverflame.deviantart.com/art/The-Shapes-Of-Hydrogen-Poster-327297786
but she could not make something nice out of it. And to echo what Robert said, please show me quantum physics related artwork that is good and scientifically accurate - a lot of it is either woo-woo a la Deepak Chopra or <a href="http://www.theguardian.com/artanddesign/jonathanjonesblog/2015/apr/23/art-respond-science-cern-ryoji-ikeda-supersymmetry">ugly modern "art"</a>.

We could drop the atom altogether, but the "nucleus and electron ellipses" is still what best evokes science, even if it is not accurate.

r-srinivas, 2016-05-06:
If you want to be a little more abstract you could maybe have a wrench superimposed on a psi letter. A pspanner, if you will.

dhslichter, 2016-05-06:
I was actually being facetious about the atomic structure -- this symbol of the ellipses definitely has its own meaning as a representation of science, regardless of what electron orbitals actually "look" like.  

I think the bicycle crank recognition has some Boulder element, but it also has to do with the fact that the wrench has a tapered width and box ends instead of open ends.  Either way, I don't have any problem if the logo stays this way.  I am just about to order a run of new crates and I put the new logo on them :)

closed 2016-05-07 by sbourdeauducq