m-labs/artiq #429: Sidepanel geometry not restored on restart

r-srinivas, 2016-05-06:
Using 1.0 RC3 on Windows, I have the side panel configured like this. 

[image]

If I close the GUI and restart, the side panel geometry is not preserved.

[image]

sbourdeauducq, 2016-05-07:
@whitequark

whitequark, 2016-05-07:
?

sbourdeauducq, 2016-05-07:
Well can you have a look into this?

whitequark, 2016-05-07:
Sure

sbourdeauducq, 2016-05-07:
#287 may be related.

r-srinivas, 2016-07-19:
Still seems to happen in 2.0.dev+485.gc0d5914  on Windows as well.

If I close the dashboard in this configuration,

[image]

it restarts as

[image]

where both the sidepanel dimensions were not restored.

sbourdeauducq, 2016-08-07:
This bug is also present on Linux with Qt 5.7.0.

sbourdeauducq, 2016-08-26:
As usual, Qt doesn't work:
https://bugreports.qt.io/browse/QTBUG-16252 (note "Priority: P1: Critical" and "Created:	22/Dec/10 10:57 AM")

The workaround https://bugreports.qt.io/browse/QTBUG-16252?focusedCommentId=281811 appears to fix this problem, at the expense of some flickering at startup.

sbourdeauducq, 2016-08-26:
This is not a reliable workaround though, as it doesn't work below a certain timer value that most probably depends on the execution speed of the system.

sbourdeauducq, 2016-08-26:
As mentioned in the Qt bug report, the problem does not manifest itself if the main window is not maximized. Resizing it to span most of the screen (without clicking the maximize button) is a manual workaround.

sbourdeauducq, 2016-08-26:
I cannot find a particular ``QEvent`` on ``QMainWindow`` we could wait for to trigger the state restore and work around this bug reliably. Seems someone has to dig deep into Qt to fix this.

sbourdeauducq, 2016-08-26:
The unreliable workaround:
```
diff --git a/artiq/frontend/artiq_dashboard.py b/artiq/frontend/artiq_dashboard.py
index 9fa2956..7088290 100755
--- a/artiq/frontend/artiq_dashboard.py
+++ b/artiq/frontend/artiq_dashboard.py
@@ -70,7 +70,9 @@ class MainWindow(QtWidgets.QMainWindow):
 
     def restore_state(self, state):
         self.restoreGeometry(QtCore.QByteArray(state["geometry"]))
-        self.restoreState(QtCore.QByteArray(state["state"]))
+        def qt_is_full_of_bugs():
+            self.restoreState(QtCore.QByteArray(state["state"]))
+        asyncio.get_event_loop().call_later(1.0, qt_is_full_of_bugs)
 
 
 class MdiArea(QtWidgets.QMdiArea):
```
Increase the 1.0 value if it does not work for you.