m-labs/artiq #430: GUI: deleting last applet from applet list closes gui

dkienzler, 2016-05-06:
Hi, 

I'm a new postdoc at NIST.

Hope this wasn't already reported.

Terminal output:
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-main\lib\site-packages\artiq\gui\applets.py", line 328, in delete
    self.applet_uids.remove(item.applet_uid)
KeyError: False

Best,
Daniel

sbourdeauducq, 2016-05-07:
The condition is not "the last applet", but I've been able to reproduce this on Linux.

sbourdeauducq, 2016-05-07:
though only partially. I had several applet UIDs incorrect set to ``False`` in my ``artiq_dashboard.pyon`` which is what was causing this bug, but I don't know how they ended up there.

dkienzler, 2016-05-07:
Thanks.

D.

jordens, 2016-05-12:
I see this as well with the browser.

```
    "AppletsDock": [(1, true, "", "$python -m artiq.applets.plot_xy --embed $ipc_address --x flopping_f_frequency flopping_f_brightness", null), (false, true, "", "$python -m artiq.applets.plot_xy --embed $ipc_address --x flopping_f_frequency flopping_f_brightness", null)]
}
```

jordens, 2016-05-25:
From #444
Delete artiq_gui.pyon and start the GUI
Right click in applets pane -> new applet
Right click in applets pane -> new applet.

jordens, 2016-05-25:
fixed in 109aa73

closed 2016-05-25 by jordens