m-labs/artiq #434: Applets not aware of missing datasets (re)appearing

jordens, 2016-05-12:
If an applet is supposed to plot a dataset (e.g. a `fit`) that does not exist yet it will never plot that dataset unless restarted. If a dataset that is deleted and recreated, the applet looses track of the dataset.
This is frequenly hit in the browser.

sbourdeauducq, 2016-05-13:
The following operations do not produce any problem:

* create "big number" applet showing ``foo`` (which doesn't exist yet)
* ``artiq_client.py set-dataset foo 34``
* ``artiq_client.py del-dataset foo``
* ``artiq_client.py set-dataset foo 42``
* create "histogram" applet showing ``quux`` (which doesn't exist yet)
* ``artiq_client.py set-dataset quux "[1,2]"``
* ``artiq_client.py del-dataset quux`` (the only slight problem is the old histogram is still displayed)
* ``artiq_client.py set-dataset quux "[4,10]"``

jordens, 2016-05-13:
Ack. Will investigate.

commit 5b5c9b2b (2016-05-13): browser: pass mods to applets (closes #434)
closed 2016-05-13 by jordens