m-labs/artiq #435: Applet processes lingering around after browser/dashboard exit/crash

jordens, 2016-05-13:
```
15495 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 12,15 flopping_f_brightness
15496 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 16,19 flopping_f_brightness
15538 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 12,15 flopping_f_brightness
15539 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 16,19 flopping_f_brightness
15580 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 12,15 flopping_f_brightness
15581 ?        Ssl    0:00 /home/rj/src/conda/envs/py35/bin/python -m artiq.applets.plot_xy --embed 16,19 flopping_f_brightness
```

I suspect they only survive when a GUI actually crashes hard. They shoud not occur in other cases.

Can we/do we want to try to clean them up better?

whitequark, 2016-05-13:
I figure the perfect solution here is to use job objects on Windows and cgroups on Linux, so that when the "root" exits, in any way whatsoever, the OS murders the rest of the processes it's grouped with. [win32api from stdlib](http://stackoverflow.com/a/12942797/254415) is the way to do this on Windows and [cgroup-utils](https://pypi.python.org/pypi/cgroup-utils) is the way to do this on Linux, although I'm not quite sure if the latter exposes all the knobs we need.

sbourdeauducq, 2016-05-13:
An easier solution without more dependencies/components may be to make sure that applets exit immediately when the other end of their IPC pipe is closed. This assumes 1) both Windows and Linux provide a sane API for this 2) Python's asyncio didn't mess it up. Those are nontrivial assumptions.

whitequark, 2016-05-13:
That doesn't handle the case where an applet is hung, and that also hangs the GUI because their message loops interlock, and the user exits the GUI process. (Although, does this actually happen with Qt window embedding? I'm pretty sure it happens with other frameworks...)

jordens, 2016-05-25:
The rationale to have child processes in new sessions is here: https://github.com/m-labs/artiq/commit/fef72506e4a25e754172a885afbc05fc15530523
It still seems correct and desirable to have this behavior for ctlmgr and workers but maybe not for applets.