m-labs/artiq #437: Change TCA6424A.set function to match documentation

r-srinivas, 2016-05-21:
In the manual, it's stated that for the LSB of the 24-bit word determines the direction of TTL0 and the MSB that of TTL23. However, if you look at the TCA6424A, [manual](http://www.ti.com/lit/ds/symlink/tca6424a.pdf), page 19, the mapping is slightly different. 

It would probably be easiest for the user if modify the set function to match the manual. There's probably a better way of doing it but I think this will work,

In artiq/artiq/coredevice/i2c.py,
```diff 
    @kernel
    def set(self, outputs):
        """Drive all pins of the chip to the levels given by the
        specified 24-bit word.
        On the QC2 hardware, the LSB of the word determines the direction of
        TTL0 (on a given FMC card) and the MSB that of TTL23.
        A bit set to 1 means the TTL is an output.
        """
+       string = format(outputs, '06x')
+       output_string = '{4}{5}{2}{3}{0}{1}'.format(string[0], string[1], string[2], string[3], string[4], string[5])
        self._write24(0x8c, 0)  # set all directions to output
+       self._write24(0x84, int(output_string, 16)  # set levels
```

sbourdeauducq, 2016-05-21:
This won't work (no string operations in kernels and the syntax isn't even correct), doing it with string format operations is hacky, and QC2-specific code does not belong in the TCA6424A driver.

Why aren't the pins connected in a simple way on the hardware anyway?

r-srinivas, 2016-05-21:
Like I said, the pins are connected to the matching TTL line. P00 - TTL0, P01 - TTL1 etc. The TCA just programmes these pins in a weird way.

sbourdeauducq, 2016-05-21:
Oh, OK. I was confused by your code that does some very unusual shuffling of 4-bit nibbles.

If I understand the datasheet correctly, the problem is you need to send the 8 LSBs of the 24-bit word first, then the middle 8 bits, then the 8 MSBs - in other words it's little-endian. Correct?

r-srinivas, 2016-05-21:
Yup.

commit 3d6f5510 (2016-05-22): coredevice/TCA6424A: convert 'outputs' value to little endian. Closes #437
commit 42c84e0c (2016-05-22): coredevice/TCA6424A: convert 'outputs' value to little endian. Closes #437
closed 2016-05-22 by sbourdeauducq