m-labs/artiq #440: support JTAG adapter of new KC705 boards

sbourdeauducq, 2016-05-23:
The USB JTAG that comes with old KC705 is an SMT-1 type, which has been discontinued. Current ones are SMT-2. To support SMT-2, one has to change which digilent source to use in the kc705.cfg in the openocd/scripts/board folder to digilent-hs2.cfg.

jordens, 2016-05-24:
@r-srinivas can you confirm that this is correct, i.e. kc705 of a certain revision (which?) have smt2 on them?

r-srinivas, 2016-05-24:
I'm not sure if they're shipped with them but we had to resolder the USB JTAG on our KC705. The SMT-1 seems to have been discontinued. 

http://store.digilentinc.com/jtag-smt1-surface-mount-programming-module/

I'm not sure if it's worthwhile to support SMT-2 within artiq, but maybe just a description in the FAQ or something would suffice.

jordens, 2016-05-24:
AFAICT kc705s do not come with SMT2. The footprint is different and the additional GPIOs on the SMT2 might actually cause problems. I would have a hard time writing a non-trivial FAQ entry for this.

closed 2016-05-24 by jordens