m-labs/artiq #442: Include pyreadline in conda package

r-srinivas, 2016-05-25:
Using 1.0 on windows, I ran into this error when trying to use the `artiq_rpctool`

```
[artiq-0524] C:\Users\rabi>artiq_rpctool -h
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-0524\Scripts\artiq_rpctool-script.py", line 3, i
n <module>
    import artiq.frontend.artiq_rpctool
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\artiq\frontend\artiq_rpct
ool.py", line 8, in <module>
    import readline  # This makes input() nicer
ImportError: No module named 'readline'
```

Installing `pyreadline` fixed this. Could this be included in the conda package?

sbourdeauducq, 2016-05-25:
How did you install pyreadline?

closed 2016-05-25 by sbourdeauducq

r-srinivas, 2016-05-25:
`conda install pyreadline` in the artiq environment. Is that okay?

jordens, 2016-05-25:
On windows yes. It's a dependency now.

sbourdeauducq, 2016-05-28:
Not anymore because platform-specific dependencies are broken with conda (of course they are). But if readline is not found now, instead of crashing you just lose line-editing capabilities in interactive mode, and a message telling you to install it.

commit 8c1f1d8f (2016-05-25): rpctool: make readline optional, add to conda dependencies. Closes #442
commit 8d0034e1 (2016-05-25): rpctool: make readline optional, add to conda dependencies. Closes #442