m-labs/artiq #443: GUI crashes when trying to close with multiple floating windows

r-srinivas, 2016-05-25:
Using 1.0 on Windows, I have the following setup across 3 monitors.

[image]

If I close the GUI, one of the floating windows sometimes persists and you see a python error, like the image below where the log window was still there after the rest of the GUI was closed,

[image]

The error message on the artiq_gui panel was,

```
c:\artiq-magtrap>call activate artiq-0524
Deactivating environment "C:\Anaconda3"...
Activating environment "C:\Anaconda3\envs\artiq-0524"...
QWindowsWindow::setGeometryDp: Unable to set geometry 442x355+0+0 on QWindow/''.
 Resulting geometry:  109x23+2560+300 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0
, 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x355+0+0 on QWindow/''.
 Resulting geometry:  109x23+2560+300 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0
, 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x355+0+0 on QWindow/''.
 Resulting geometry:  109x23+2560+300 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0
, 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x355+0+0 on QWindow/''.
 Resulting geometry:  109x23+2560+300 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0
, 0, minimum size: 0x0, maximum size: 16777215x16777215).
External WM_DESTROY received for  QWidgetWindow(0x40cbb80, name = "DDSWindow") ,
 parent:  QObject(0x0) , transient parent:  QWidgetWindow(0x40cc060, name = "Mai
nWindowClassWindow")
External WM_DESTROY received for  QWidgetWindow(0x40cbdc0, name = "TTLWindow") ,
 parent:  QObject(0x0) , transient parent:  QWidgetWindow(0x40cc060, name = "Mai
nWindowClassWindow")
[artiq-0524] c:\artiq-magtrap>
```

Once when the DDS panel persisted,

```
c:\artiq-magtrap>call activate artiq-0524
Deactivating environment "C:\Anaconda3"...
Activating environment "C:\Anaconda3\envs\artiq-0524"...
QWindowsWindow::setGeometryDp: Unable to set geometry 442x280+0+0 on QWindow/''.
 Resulting geometry:  109x23+640+280 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0,
 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x280+0+0 on QWindow/''.
 Resulting geometry:  109x23+640+280 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0,
 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x280+0+0 on QWindow/''.
 Resulting geometry:  109x23+640+280 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0,
 0, minimum size: 0x0, maximum size: 16777215x16777215).
QWindowsWindow::setGeometryDp: Unable to set geometry 442x280+0+0 on QWindow/''.
 Resulting geometry:  109x23+640+280 (frame: 0, 0, 0, 0, custom margin: 0, 0, 0,
 0, minimum size: 0x0, maximum size: 16777215x16777215).
External WM_DESTROY received for  QWidgetWindow(0x8091ac0, name = "log0Window")
, parent:  QObject(0x0) , transient parent:  QWidgetWindow(0x41454f0, name = "Ma
inWindowClassWindow")
External WM_DESTROY received for  QWidgetWindow(0x4145250, name = "TTLWindow") ,
 parent:  QObject(0x0) , transient parent:  QWidgetWindow(0x41454f0, name = "Mai
nWindowClassWindow")
[artiq-0524] c:\artiq-magtrap>
```

I can't reproduce this super reliably but this seems to happen about once every 3-4 times I try. Seems like sometimes there's an error getting all the floating windows to close.

jordens, 2016-05-25:
What's in that error dialog (the screenshots are not readable)? Which of those messages appear during the closing?

r-srinivas, 2016-05-25:
That error diologue was that Python had stopped working. From the screen shot was the first error message. Once the dds_panel didn't close and that had produced the second error message.

jordens, 2016-05-25:
There are multiple messages in each snippet you pasted. Did all messages appear when closing the main application window or were some of them there before?

r-srinivas, 2016-05-25:
I pasted all the messages that appeared on the artiq_gui command window. The unable to set geometry messages are usually there with the configuration we have. The messages that appeared upon closing were the `External WM_Destroy` ones. Somehow that message doesn't appear for the window that persists and causes the crash. There weren't any messages on on the artiq-master window.

jordens, 2016-05-25:
Are the WM_DESTROY messages not emitted if the GUI closes cleanly, that is without the "Python has stopped working" dialog showing up? To keep the window and the messages from disappearing in that case, rung `artiq_gui` from a console, windows cmd or conda python shell.

jordens, 2016-05-25:
May or may not be the same as #445

sbourdeauducq, 2016-05-25:
I guess this is the usual PyQt memory management bug.

r-srinivas, 2016-05-25:
You're right Robert, the WM_Destroy messages are also emitted if the GUI closes cleanly. What's the best way of obtaining a backtrace?

Also, it seems that this error is harder to produce once I had closed about 6 of the Vaunix Attenuator GUIs, so perhaps it is a memory management thing.

sbourdeauducq, 2016-05-25:
If you have Visual Studio installed, you should have a "Debug the program" option that appears in the "Python has stopped working" window. You should be able to get a backtrace from that.

What is that desktop background?

sbourdeauducq, 2016-07-05:
Does it still crash after 11930228fd6b8a2445f3ae4a9729568bfd2c2ab6?

r-srinivas, 2016-07-05:
I don't have the dev installation right now but I patched artiq\frontend\artiq_gui.py on 1.1 and I can't really reproduce it. I'll close it for now and comment here if I run into this again.

closed 2016-07-05 by r-srinivas