m-labs/artiq #445: segfault when closing the dashboard

jordens, 2016-05-25:
* Delete `artiq_dashboard.pyon`
* Start dashboard
* In Applets: RMB->New
* In Applets: RMB->New
* Close dashboard
* Crash (sometimes)

This is with current master (109aa73), conda, conda qt5 etc.

```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007ffff30fc289 in qpycore_qobject_metaobject(_sipSimpleWrapper*, _sipTypeDef*)
    () from /home/rj/src/conda/envs/py35/lib/python3.5/site-packages/PyQt5/QtCore.so
(gdb) bt
#0  0x00007ffff30fc289 in qpycore_qobject_metaobject(_sipSimpleWrapper*, _sipTypeDef*) () from /home/rj/src/conda/envs/py35/lib/python3.5/site-packages/PyQt5/QtCore.so
#1  0x00007ffff2c3c1f8 in QObject::~QObject() () from /home/rj/src/conda/envs/py35/lib/python3.5/site-packages/PyQt5/../../../libQt5Core.so.5
#2  0x00007ffff1927478 in ?? () from /home/rj/src/conda/envs/py35/lib/python3.5/site-packages/PyQt5/../../../libQt5Gui.so.5
#3  0x00007ffff6a23fe8 in __run_exit_handlers (status=0, listp=0x7ffff6dad5f8 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true) at exit.c:82
#4  0x00007ffff6a24035 in __GI_exit (status=<optimized out>) at exit.c:104
#5  0x00007ffff6a0a837 in __libc_start_main (main=0x400980 <main>, argc=2, argv=0x7fffffffdb38, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>,
    stack_end=0x7fffffffdb28) at ../csu/libc-start.c:325
#6  0x00000000004008b9 in _start ()
(gdb)
```

sbourdeauducq, 2016-05-25:
http://stackoverflow.com/questions/11945183/what-are-good-practices-for-avoiding-crashes-hangs-in-pyqt
"The easiest way to avoid exit crashes is to call os._exit() before python starts collecting Qt objects" ...

sbourdeauducq, 2016-07-01:
@jordens Do you still see those exit crashes after 11930228fd6b8a2445f3ae4a9729568bfd2c2ab6?

jordens, 2016-07-04:
No. But reverting (1193022) doesn't reproduce them. Nor does the original (109aa73) crash here. Let's close. I'll reopen if I see them again.

closed 2016-07-04 by jordens