m-labs/artiq #446: Input pulse counting is slow

dhslichter, 2016-05-26:
Running tests to determine the amount of time required for `TTLInOut.count()`, it seems to take roughly 3000-5000 mu to complete (30-50 us at 100 MHz RTIO clock).  This is diagnosed by running the experiment below and checking the slack listed on the RTIOUnderflow error.  Run-to-run variations have this slack go as low as ~2000 mu and as high as ~6000 mu shot to shot.  

This number appears to be fairly independent of the duration of the gating window `tpulse` and thus the number of counts.  Our setup has about 2 events in a 10 us window or 200 events in a 1 ms window, and the underflow is indistinguishable between these cases.  

```python
class CountTest(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")      
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_pmt")
        
    def run(self):       
        tpulse = int(1000, width=64)
        self.pulse(tpulse)
                
    @kernel
    def pulse(self, tpulse):

        with parallel:
            self.ttl_bd1.pulse_mu(tpulse)                
            self.ttl_bd2.pulse_mu(tpulse) 
            self.ttl_pmt.gate_rising_mu(tpulse)
        self.ttl_pmt.count()
        self.ttl_bd1.pulse_mu(tpulse)
```

With a separate experiment to pull the individual timestamps (see kernel below), the time required is similar.  

```python
    @kernel
    def pulse(self, tpulse):

        with parallel:
            self.ttl_bd1.pulse_mu(tpulse)                
            self.ttl_bd2.pulse_mu(tpulse) 
            self.ttl_pmt.gate_rising_mu(tpulse)
        while True:
            ts = self.ttl_pmt.timestamp_mu()
            if ts == -1:
                break
        self.ttl_bd1.pulse_mu(tpulse)
```

Why is this so slow, and can it be optimized in any way to reduce this dead time?

jordens, 2016-05-26:
With that method the number you quote also includes the time to do the ttl_bd1.on() after count().
The work that needs to be done in count() is dominated by reading the rtio fifo status (looping), then the timestamp, acknowledging the event and comparing with the gate-closure time, then looping over that.

How fast did you expect it to be? And what do you mean by dead time? count() will obviously eat up all slack. You can make good use of that time.by pushing events (e.g. ttl_bd1.pulse_mu()) for times after the gate closure, before doing count().

dhslichter, 2016-05-26:
Not sure what you mean by `ttl_bd1.on()`; are you referring to the final pulse `ttl_bd1.pulse_mu(tpulse)` at the end of the kernel?  This time is not included in the quoted slack; changing the duration of this final pulse (from 1000 to 10000, for example) has no impact on the reported slack as far as I can see.  The point is that `count()` takes 30-50 us (sometimes more) to execute.  In fact, I am now re-running the identical code I have pasted above, and the underflow slack is now ~8000-12000 mu, thus 80-120 us.  This kind of variability is problematic if you are hard-coding delays into your kernel, because then you need the delays to be ~ms-scale to ensure you'll be OK.  

It is definitely possible to try scheduling other pulses (e.g. long recooling pulses) before `count()`, such that the time spent getting information out of the FIFO is being used profitably by the rest of the experiment in the meantime, and we have already refactored our Ion Monitor code to do this, for example.  

However, it would be nice to see if there is any low-hanging fruit to speed up this process.  In addition, if one wants to do feedback or fast branching, this time (and the required delay, which must be longer than the longest case time that `count()` will take or else you get an underflow and your experiment quits) is the limiting factor on your ability to do so.  

In addition, this kind of behavior should be clearly documented in the description of the `count()` method in the manual, because 3000-5000 mu is a nontrivial time restriction that users should be aware of.  Perhaps adding a reminder that since the events are in the FIFO, one can/should wait to call `count()` until some part of the experiment where things aren't happening so fast.  In particular, the fact that calling `count()` doesn't simply advance the timeline to when the counting is done, but requires an appropriate delay or other pulse to give it headroom in the timeline, is a subtle behavior that would be very good to spell out more clearly.

jordens, 2016-05-27:
> Not sure what you mean by ttl_bd1.on(); are you referring to the final pulse ttl_bd1.pulse_mu(tpulse) at the end of the kernel?  This time is not included in the quoted slack;

`pulse_mu()` starts with an `on()`. That `on()` is what triggers the Underflow! And all the code leading to it is included in the amount of slack violation.

> The point is that count()  takes 30-50 us (sometimes more) to execute. In fact, I am now re-running the identical code I have pasted above, and the underflow slack is now ~8000-12000 mu, thus 80-120 us. 

1 mu is 1 ns here. 3000 mu are 3 µs.

> This kind of variability is problematic if you are hard-coding delays into your kernel, because then you need the delays to be ~ms-scale to ensure you'll be OK.

What is problematic now, the variability or the value?

> In addition, this kind of behavior should be clearly documented in the description of the count() method in the manual, because 3000-5000 mu is a nontrivial time restriction that users should be aware of.

What would be a trivial time restriction?
I am still not sure how fast did you expect it to be and what you mean by dead time.

jordens, 2016-05-27:
One possible optimization would be to feed back the sensitivity events into the input fifo. Then count() would not need to read and compare the current rtio counter with the memorized gate closure and can also return a bit earlier. But that would mean rewriting quite a bit of code and it won't yield a lot of speed up. The other way would be something analogous to the frequency output ttl: a counter input ttl. But it also would not make things much faster for a handful of input events.

And again those 3-5µs include latency that is unrelated to `count()`.

dhslichter, 2016-05-27:
Ack, forgot that mu represents the fine RTIO timestamp (1.25 ns in our case with 100 MHz RTIO clock), not the coarse one.  In this case, ~10 us for a `count()` event is satisfactory for our purposes now.

closed 2016-05-27 by dhslichter