m-labs/artiq #448: Argument units in GUI not passed to experiment

dhslichter, 2016-05-27:
If units are specified in a `NumberValue()`, as in the code below, the value of the argument when referenced in the code should also contain these units.  For example, the code below should print 1e-4, when in fact it prints 100.0 instead.  Currently, one needs to manually multiply by `us` every time `self.t_detect` is referenced in the code.

```python
    def build(self):
        self.setattr_device("core")
        self.setattr_argument("t_detect", NumberValue(100, unit="us", 
            ndecimals=1, step=0.1))

    @kernel
    def run(self):
        print(self.t_detect)
```

jordens, 2016-05-27:
`unit` is decorative. add `scale = 1*us`.

sbourdeauducq, 2016-05-27:
In 2.0, we could easily change the default of ``scale`` from ``1.0`` to ``None``, and a ``None`` scale would mean "guess from ``unit`` based on a simple dictionary". Mostly backwards-compatible.

dhslichter, 2016-05-27:
Thanks.  It would be good to describe this more clearly in the documentation at the very least.  I would rephrase/expand the general description of `NumberValue()` to include an example of how to do unit/scale, e.g.:

For arguments with units, use both the unit parameter (a string for display) and the scale parameter (a numerical scale for experiments).  For example, `NumberValue(1, unit="ms", scale=1*ms)` will display as 1 ms in the GUI window because of the unit setting, and appear as the numerical value 0.001 in the code because of the scale setting.

parameters:
unit - a string representing the unit of the value, for display purposes only
scale - a numerical scaling factor by which the displayed value is multiplied when referenced in the experiment.

dhslichter, 2016-05-27:
And yes, @sbourdeauducq if one can do this kind of reverse lookup it would be nice as well.

sbourdeauducq, 2016-05-27:
The rationale for not having ``scale`` determined by ``unit`` is one may potentially use complex units for which automatic scale determination is difficult or impossible. But a dictionary of simple units and a clear error message telling the user that they need to specify the scale manually would be OK too.

dhslichter, 2016-05-27:
I understand and can appreciate this rationale, but if a simple dictionary plus an error message saying "specify scale" can be implemented I think it will make most use cases vastly easier.

Also, in my example above, it should read `NumberValue(1*ms, unit="ms", scale=1*ms)`

sbourdeauducq, 2016-05-27:
Yes, that's very straightforward.

commit 0faa2d56 (2016-05-28): language/environment: autoset scale for common units. Closes #448
closed 2016-05-28 by sbourdeauducq