m-labs/artiq #449: Pulses longer than 200 ms do not end

r-srinivas, 2016-05-27:
Running the code below, setting `t` to be anything longer than 0.2 results in `ttl15` staying high indefinitely.  

```python
class IonLoad2(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl15")
        self.setattr_argument("t",NumberValue(1., step=0.05, ndecimals=2))

    @kernel
    def run(self):
        self.ttl15.off()
        delay_mu(1000)
        self.ttl15.pulse(self.t)
```

jordens, 2016-05-27:
The pulse is working fine. The RTIO FIFOs are reset by the next exeriment. #391
You can `ttl15.sync()`.

closed 2016-05-27 by jordens

r-srinivas, 2016-05-27:
If there is no experiment after it doesn't turn off. `self.ttl15.sync()` fixes it, but do we need to call it for every pulse then?

jordens, 2016-05-27:
There is the idle kernel.

No. Until that hidden and undesirable resetting of the FIFOs is fixed, you have to make sure that the pulses pushed into the FIFO are executed. Just sync() (or a gate and count()) at the end of the experiment.

Looks like that basic tutorial/description of how ARTIQ RTIO works and how experiments are executed is desparately needed.

dhslichter, 2016-05-27:
Again, not well documented anywhere unless you like reading through old closed issues.  The documentation for `artiq.coredevice.ttl()` should explicitly state that the kernel will load the FIFOs and exit, and if a new experiment comes in (including idle kernel).

dhslichter, 2016-05-27:
@jordens yes, a good description of how the kernel "thinks" and what the process is for an experiment executing will save you from being asked all these same questions yet again for each new user who starts up with ARTIQ :)

jordens, 2016-05-27:
Paraphrasing what others have said and meta-diagnosing is not very helpful.

Your suggestion to extend the API documentation doesn't cut it. Describing the concepts there would be wrong.

Are you guys helping each other locally a bit? That might avoid some of this ping pong until the documentation is perfect.

sbourdeauducq, 2016-05-28:
> you have to make sure that the pulses pushed into the FIFO are executed. Just sync() (or a gate and count()) at the end of the experiment.

Or wait until core.rtio_counter() is above now(), as I pointed out in other issues.