m-labs/artiq #452: Incorrect error message when trying to access get_dataset in kernel

r-srinivas, 2016-05-27:
Using 1.0 on Windows, I have the following experiment,

```python
from artiq.experiment import *
from repository.magtrap_subcomponents import Detection
import numpy as np

class BD_Scan_Test_2(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_bd_ne")
        self.setattr_device("dds_bd1")
        self.setattr_device("scheduler")
        self.detect = Detection(parent=self)
        
        self.setattr_argument("frequency_scan", Scannable(
            default=LinearScan(300, 340, 100)))
        self.setattr_argument("repetitions", NumberValue(100, min=10, max=500, ndecimals=0))
        self.setattr_argument("nbins", NumberValue(100, ndecimals=0))

    def run(self):
        l = len(self.frequency_scan)
        self.set_dataset("dds_frequency", np.full(l, np.nan), 
                         broadcast=True, save=True)
        self.set_dataset("counts", np.full(l, np.nan),
                         broadcast=True, save=True)
        self.counts = [0]*int(self.repetitions)
        index = []        
        frequencies = []        
        for i, f in enumerate(self.frequency_scan):
            index.append(i)            
            frequencies.append(f)
        self.experiment(index, frequencies)
    
    @kernel
    def experiment(self, index, frequencies):
        for i in index:
            self.dds_scan(i, frequencies[i])
        self.dds_bd1.set(self.get_dataset("dds_f_bd1"))
    
    @kernel
    def dds_scan(self, index, frequency):
        hist = [0]*int(self.repetitions)
        total = 0
        self.core.break_realtime()
        self.dds_bd1.set(frequency*MHz)
        #self.ttl_bdd.off()
        for i in range(int(self.repetitions)):
            with parallel:
                self.ttl_bd1.pulse(50*us)
                self.ttl_bd2.pulse(50*us)
                self.ttl_bd_ne.pulse(50*us)
            self.ttl_bdd.pulse(500*us)
            n = self.detect.bd_ne(100*us)
            delay(1*ms)
            if n>= int(self.nbins):
                n = int(self.nbins)-1
            hist[n] += 1
            total += n
        delay(100*us)
        self.ttl_bdd.on()
        self.set_dataset("photon_histogram", hist, broadcast=True)
        self.mutate_dataset("counts", index, total)
        self.mutate_dataset("dds_frequency", index, frequency)    
```

Which raises the error,

```
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1012
ERROR:worker(1013,bd_scan_test_2.py):root:While compiling <repository>\magtrap\b
d_scan_test_2.py
<repository>\magtrap\bd_scan_test_2.py:54:26-54:39: error: cannot unify float wi
th NoneType
        self.dds_bd1.set(frequency*MHz)
                         ^^^^^^^^^^^^^
<repository>\magtrap\bd_scan_test_2.py:54:26-54:39: note: expression of type flo
at
        self.dds_bd1.set(frequency*MHz)
                         ^^^^^^^^^^^^^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 1013
```

However, the error is in this line here,

```python
 self.dds_bd1.set(self.get_dataset("dds_f_bd1"))
```

If I get rid of it, it's fine.

sbourdeauducq, 2016-05-28:
Again, RPCd functions that return something else than None must have an appropriate type annotation (and therefore you cannot call ``get_dataset`` directly in a kernel).
https://m-labs.hk/artiq/manual-release-1/compiler.html#remote-procedure-calls

sbourdeauducq, 2016-05-28:
The compiler should have pointed out that line though.

r-srinivas, 2016-05-28:
Sorry, I should have been clearer. I get why I get the error message, but the error message points me to the wrong line.

whitequark, 2016-05-28:
@sbourdeauducq There is no way to fix this with unification-based inference. It is not directional. Inferring the type of the argument to `TTLInOut.set` as None from line `self.dds_bd1.set(self.get_dataset("dds_f_bd1"))` is just as valid as inferring it as float from line `self.dds_bd1.set(frequency*MHz)`.

However, what we can do is add explicit type annotations on every function in our standard library, which will considerably enhance the quality of error messages.

sbourdeauducq, 2016-05-28:
Would it be difficult to display where the types were inferred in error messages?

whitequark, 2016-05-28:
It's not completely trivial (there are very many corner cases that need to be tested, so it is a lot of work to add this), and more importantly, it's not going to give very good results; most of the information I can collect during unification is irrelevant, so this wouldn't necessarily make your life easier. (It would on trivial examples such as this one, but not once the length of the path inference takes grows.)

This is basically the reason most MLs require explicit specification of types on module or even function boundaries.

whitequark, 2016-05-28:
I'll look up some papers on this problem later, hopefully someone thought up a decent way...

whitequark, 2017-10-02:
Triage: fixing this would be a research project that I don't think has a low enough cost/benefit ratio to undertake.

closed 2017-10-02 by whitequark