m-labs/artiq #453: Object dtype dtype ('O') has no native HDF5 equivalent

r-srinivas, 2016-05-28:
Using 1.0 on Windows, if I run this experiment, it runs fine and it plots the data but I get an error message at the end.

```python
from artiq.experiment import *
from repository.magtrap_subcomponents import Detection
import numpy as np

class BD_Scan_Test_2(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_bd_ne")
        self.setattr_device("dds_bd1")
        self.setattr_device("scheduler")
        self.detect = Detection(parent=self)
        
        self.setattr_argument("frequency_scan", Scannable(
            default=LinearScan(300, 340, 100)))
        self.setattr_argument("repetitions", NumberValue(100, min=10, max=500, ndecimals=0))
        self.setattr_argument("nbins", NumberValue(100, ndecimals=0))
        self.f0 = self.get_dataset("dds_f_bd1")

    def run(self):
        l = len(self.frequency_scan)
        self.set_dataset("dds_frequency", np.full(l, np.nan), 
                         broadcast=True, save=True)
        self.set_dataset("counts", np.full(l, np.nan),
                         broadcast=True, save=True)
        self.counts = [0]*int(self.repetitions)
        index = []        
        frequencies = []        
        for i, f in enumerate(self.frequency_scan):
            index.append(i)            
            frequencies.append(f)
        self.experiment(index, frequencies)
        
    
    @kernel
    def experiment(self, index, frequencies):
        for i in index:
            self.dds_scan(i, frequencies[i])
    
    @kernel
    def dds_scan(self, index, frequency):
        hist = [0]*int(self.repetitions)
        total = 0
        self.core.break_realtime()
        self.dds_bd1.set(frequency*MHz)
        #self.ttl_bdd.off()
        for i in range(int(self.repetitions)):
            with parallel:
                self.ttl_bd1.pulse(50*us)
                self.ttl_bd2.pulse(50*us)
                self.ttl_bd_ne.pulse(50*us)
            self.ttl_bdd.pulse(500*us)
            n = self.detect.bd_ne(100*us)
            delay(1*ms)
            if n>= int(self.nbins):
                n = int(self.nbins)-1
            hist[n] += 1
            total += n
        delay(100*us)
        self.ttl_bdd.on()
        self.set_dataset("photon_histogram", hist, broadcast=True)
        self.mutate_dataset("counts", index, total)
        self.mutate_dataset("dds_frequency", index, frequency)  
```

```
ERROR:worker(1034,bd_scan_test_2.py):root:Terminating with exception (TypeError:
 Object dtype dtype('O') has no native HDF5 equivalent)
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\artiq\master\worker_impl.
py", line 234, in main
    dataset_mgr.write_hdf5(f.create_group("datasets"))
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\artiq\master\worker_db.py
", line 182, in write_hdf5
    f[k] = v
  File "h5py\_objects.pyx", line 54, in h5py._objects.with_phil.wrapper (C:\Mino
nda\conda-bld\work\h5py\_objects.c:2696)
  File "h5py\_objects.pyx", line 55, in h5py._objects.with_phil.wrapper (C:\Mino
nda\conda-bld\work\h5py\_objects.c:2654)
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\h5py\_hl\group.py", line
292, in __setitem__
    ds = self.create_dataset(None, data=obj, dtype=base.guess_dtype(obj))
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\h5py\_hl\group.py", line
105, in create_dataset
    dsid = dataset.make_new_dset(self, shape, dtype, data, **kwds)
  File "C:\Anaconda3\envs\artiq-0524\lib\site-packages\h5py\_hl\dataset.py", lin
e 93, in make_new_dset
    tid = h5t.py_create(dtype, logical=1)
  File "h5py\h5t.pyx", line 1450, in h5py.h5t.py_create (C:\Minonda\conda-bld\wo
rk\h5py\h5t.c:16090)
  File "h5py\h5t.pyx", line 1470, in h5py.h5t.py_create (C:\Minonda\conda-bld\wo
rk\h5py\h5t.c:15924)
  File "h5py\h5t.pyx", line 1525, in h5py.h5t.py_create (C:\Minonda\conda-bld\wo
rk\h5py\h5t.c:15825)
TypeError: Object dtype dtype('O') has no native HDF5 equivalent
ERROR:master:artiq.master.scheduler:got worker exception in analyze stage, delet
ing RID 1034
```

Am I doing something wrong with the datasets?

sbourdeauducq, 2016-05-28:
Can you narrow down which dataset causes the error?

sbourdeauducq, 2016-05-28:
Shorter repro:
```
>>> import h5py
>>> x = h5py.File("oo.h5", "w")
>>> from artiq.experiment import *
>>> x["oopp"] = [int(4)]
```

sbourdeauducq, 2016-05-28:
np.array of a list of redefined ints also gives the wrong result.

@r-srinivas As a workaround, write an intermediate function that sets ``photon_histogram`` passing the data through ``np.array([host_int(v) for v in hist])``

sbourdeauducq, 2016-05-28:
@whitequark Is there any way this can be fixed other than #424?

whitequark, 2016-05-28:
@sbourdeauducq sure. we can make worker_db (or some better place, perhaps) recognize and convert ARTIQ ints into proper ints.

sbourdeauducq, 2016-05-28:
meh, that will still break when the user does things like ``np.array(list_of_artiq_ints)``.

whitequark, 2016-05-28:
Yes, you need a deep map.

This is arguably a bug in hdf5, because `host_int` defines a conversion to regular `int`:
```
    def __int__(self):
        return self._value
```

sbourdeauducq, 2016-05-28:
> Yes, you need a deep map.

Yes, but even that is a half-baked solution. ``np.array(list_of_artiq_ints)`` is valid user code, and I reckon it will even be common until #424 is done.

whitequark, 2016-05-28:
Okay, then we need to patch hdf5. I can look into it.

sbourdeauducq, 2016-05-28:
Your int redefinition does seem to have unusual effects. hdf5 and numpy have problems with it, but can you really blame them? Won't similar issues creep up with other libraries?
```
>>> from artiq.experiment import *
>>> isinstance(3, int)
False
```

whitequark, 2016-05-28:
That's completely correct. It is emphatically not Python's infinite width int, and under no circumstances should be treated as such. It can be *converted* to one though, and these libraries ought to use the interface that Python provides for such conversion.

sbourdeauducq, 2016-05-28:
Maybe it should not be called ``int`` then.

whitequark, 2016-05-28:
We can rename it to something else (`fixed`?), which will need to be done in both host code and ARTIQ Python. I don't like that it's called `int` either.

sbourdeauducq, 2016-05-28:
I'm for replacing ARTIQ int with the actual NumPy types:

1. They are already there.
2. NumPy is well-established so other libraries have had more opportunity to fix whatever specific issues they could have with them.
3. They even behave almost as we want on the host:
```
>>> x = np.int32(2**31-1)
>>> x + 1
2147483648
>>> type(x + 1)
<class 'numpy.int64'>
>>> type(x)
<class 'numpy.int32'>
>>> x = np.int64(2**63-1)
>>> type(x)
<class 'numpy.int64'>
>>> x + 1
__main__:1: RuntimeWarning: overflow encountered in long_scalars
-9223372036854775808
```

whitequark, 2016-05-28:
> almost

Well, here's your problem. This will just result in inconsistencies in portable code.

whitequark, 2016-05-28:
Although we could enable overflow checking in OR32 and kill that problem before it appears.

sbourdeauducq, 2016-05-28:
The inconsistencies is not worse than with the current ARTIQ int IMO. Yes, overflow checking would be nice but low-priority.

whitequark, 2016-05-28:
Okay, I'll look further into it.

sbourdeauducq, 2016-05-28:
Why did you redefine ``round`` by the way?
```python
def round(value, width=32):
    return int(host_round(value), width)
```
AFAICT doing this in ``int.__round__`` would have been enough. And numpy already does it that way:

```
>>> type(round(np.int32(4)))
<class 'numpy.int32'>
>>> round
<built-in function round>
```

whitequark, 2016-06-20:
Also depends on #408.

commit 906db876 (2016-07-06): language: replace coredevice int with numpy.{int32,int64}.

Fixes #453.
closed 2016-07-06 by whitequark