m-labs/artiq #454: browser cannot open experiments that access devices

sbourdeauducq, 2016-05-28:
```
ERROR:worker(None,blink_forever.py):root:Terminating with exception (DeviceError: Failed to create device 'core')
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/master/worker_db.py", line 154, in get
    dev = _create_device(desc, self)
  File "/home/sb/artiq/artiq/master/worker_db.py", line 88, in _create_device
    ty = desc["type"]
TypeError: 'NoneType' object is not subscriptable

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/sb/artiq/artiq/master/worker_impl.py", line 218, in main
    exp_inst = exp((device_mgr, dataset_mgr, argument_mgr))
  File "/home/sb/artiq/artiq/language/environment.py", line 186, in __init__
    self.build(*args, **kwargs)
  File "/home/sb/artiq/artiq/examples/master/repository/coredevice_examples/simple/blink_forever.py", line 6, in build
    self.setattr_device("core")
  File "/home/sb/artiq/artiq/language/environment.py", line 243, in setattr_device
    setattr(self, key, self.get_device(key))
  File "/home/sb/artiq/artiq/language/environment.py", line 236, in get_device
    return self.__device_mgr.get(key)
  File "/home/sb/artiq/artiq/master/worker_db.py", line 157, in get
    .format(name)) from e
artiq.master.worker_db.DeviceError: Failed to create device 'core'
```

sbourdeauducq, 2016-07-06:
@jordens ping

closed 2016-07-07 by jordens

sbourdeauducq, 2019-10-09:
That's not the same error message and likely a user error, please use the forum https://forum.m-labs.hk
Please provide the full error message and your device_db file.

sbourdeauducq, 2019-10-09:
Please see the AD9910 datasheet on how to configure the PLL.

jordens, 2019-10-09:
Also please don't hijack issues. It's not helpful. Your problem is unrelated.

commit 2a5a1f32 (2016-07-07): browser, worker: feed experiments dummy devices, closes #454

* just returning `None` as dummy device (like ExamineDeviceMgr)
is not explicit enough, certainly hard to debug
* introducing a special flag for the `build` action does not
seem the right place