m-labs/artiq #460: scientific spinbox

sbourdeauducq, 2016-06-01:
See discussion in #457 

Other points:
* dealing with ``step``
* using ``ScientificSpinBox`` for non-integer ``NumberValue``

sbourdeauducq, 2016-06-01:
Regarding ``step``:

@jordens Your scientific spinbox design modifies the coefficient part by a fixed amount (whatever the exponent) when the arrows are clicked. ARTIQ/Qt has a ``step`` parameter that modifies the floating point value by a fixed amount. What shall we do?

jordens, 2016-06-01:
I had put a couple thoughts on that general topic in comments in the code in the scanwidget. I would like to consistently link the four sources of "step/resolution/tick/snap": the ticker step, the spinboxes value resolution, the spinbox step size, the ticker step modified by npoints. But I have not decided yet what the best approach is and whether it is actually doable.

jordens, 2016-06-21:
The current implementation looses the unit suffix most of the time when ctrl- or shift-wheeling on a spinbox.

jordens, 2016-08-11:
An intuitive coupling of `singleStep` and `relativeStep` with the ticker step is still desirable but not urgent.

jordens, 2016-08-11:
Forked issue.

commit 6fe23b88 (2016-08-11): gui: new ScientificSpinBox (from 09f9293) closes #460
closed 2016-08-11 by jordens