m-labs/artiq #463: scan widget colored handle size too small

jboulder, 2016-06-03:
ARTIQ 1.0 on Windows

Upper and lower scan bounds handles have colored markers. The markers are too small to see.

sbourdeauducq, 2016-06-03:
only happens sometimes such as some UMD PCs.

jordens, 2016-06-03:
Those sizes are derived the same way as e.g. the default window sizes. Are those also too small?
Is this some "theme" issue (the scientific spinbox etc not seeing the same themed font)?

sbourdeauducq, 2016-06-04:
It seems like the actual size is correct, but their bottom part is cut because the widget is not large enough.

jordens, 2016-06-04:
Got a screenshot?

sbourdeauducq, 2016-06-06:
Could be related to hi-DPI (that's on a Windows Surface).
[image]

jordens, 2016-08-04:
@jboulder  can you still reproduce this? The scientific spinboxes were dropped after 1.0 and look very different than the other spinboxes in the screenshot. Do you have access to artiq on a windows surface? OK to defer this issue till after 2.0 or is it actively hurting somebody right now?

jbqubit, 2016-08-04:
I don't currently have access to a high-dpi display like a Microsoft
surface.

jordens, 2017-01-27:
timed out

closed 2017-01-27 by jordens

jbqubit, 2017-01-27:
Thanks for the reminder. I now have access to a Windows 10 machine with a high-resolution display (2560x1140). The scan widget and spin boxes are sized correctly and handles are easy to operate.

[image]

In a multi-display setup I can also have the ARTIQ Dashboard lying half on a high-resolution display and half on a low-resolution display and everthing is rendered correctly. Nice.