m-labs/artiq #482: Automatically bit shift write data based on MSB/LSB

r-srinivas, 2016-06-14:
Looking at the SPI code for the AD5360,

```python
 @kernel
    def write(self, data):
        """Write 24 bits of data.

        This method advances the timeline by the duration of the SPI transfer
        and the required CS high time.
        """
        self.bus.write(data << 8)
        delay_mu(self.bus.ref_period_mu)  # get to 20ns min cs high
```

You have to bitshift the data by 8 as only 24 bits are written, but the write register is 32 bit. However, this is only true for MSB programming. For LSB programming this bit shift is not necessary. It would be nice if this were handled automatically based on the LSB config value. If not, this should be added to the documentation, maybe under the spi write function.

jordens, 2016-06-14:
That shifting is much better done in software.
The behavior is documented:
https://github.com/m-labs/artiq/blob/master/artiq/coredevice/spi.py#L78-L82
When you write a driver on top of `spi`, you have read that docstring. I am hesitant to rip that part out because it is here where the size/lsb/alignment behavior topic is described as a whole.

r-srinivas, 2016-06-14:
It could be helpful to include that under write, something like,

````
For MSB programming, write data needs to be shifted to 32 bits.
seealso:: :meth:`set_config`
```

commit a8b211f8 (2016-06-15): spi: cross-reference bit ordering and alignment, closes #482
commit 35895283 (2016-06-15): spi: cross-reference bit ordering and alignment, closes #482
closed 2016-06-15 by jordens