m-labs/artiq #487: Construct dependent arguments

r-srinivas, 2016-06-24:
Is it possible to construct dependent arguments? For example,

```python

from artiq.experiment import *

class Data_Label_test(EnvExperiment):

    def build(self):
        self.setattr_argument('test1', NumberValue(1), 'Category 1')
        self.setattr_argument('test2', NumberValue(self.test1), 'Category 2')

    def run(self):
        pass
```

If I change the value of test 1 and try to recompute test2, it doesn't update.

sbourdeauducq, 2016-06-27:
Not possible with the current code.

When do you expect it to update?

r-srinivas, 2016-06-28:
I was hoping that clicking on recompute argument would retake the updated argument from `test1`.

sbourdeauducq, 2016-06-29:
That wouldn't be a great design, because the order of the widgets is the order of the calls. You could have a value depend only on the value of widgets above it, which sounds arbitrary.

sbourdeauducq, 2016-06-29:
Why do you need to have such dependencies?

r-srinivas, 2016-06-29:
The experiment I wanted to run was this,

```python
# -*- coding: utf-8 -*-
"""
BD Scan Test
All experiments and scan points on core device so no delay between kernels
"""

from artiq.experiment import *
from repository.magtrap_subcomponents import Detection
from repository.fit_functions import Rabi_Flopping
import numpy as np

class MW_3p3_2p2_Scan(EnvExperiment):
    '''MW 3p3 2p2 Time Scan'''

    def build(self):
        #  build hardware for experiment
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_bd1")
        self.setattr_device("ttl_bd2")
        self.setattr_device("ttl_bd_ne")
        self.setattr_device("dds_box")
        self.setattr_device("ttl_ps0")
        self.setattr_device("ttl_mw_mux")
        self.setattr_device("ttl_mw_ab")
        self.setattr_device("dds_bd1")
        #  instance of detection subcomponent
        self.detect = Detection(parent=self)

        #  arguments in gui
        self.f0 = self.get_dataset("mw.f.3p3_2p2")/(1e6)
        self.setattr_argument("dds_bd1_frequency",
                              NumberValue(self.get_dataset('dds_f.bd1') - 6e6,
                                          scale=1e6, unit='MHz'))
        self.setattr_argument("frequency_scan", Scannable(
            default=NoScan(self.f0), unit='MHz'))
        self.setattr_argument("time_scan", Scannable(
            default=LinearScan(0*us, 10*us, 100), scale=1e-6, unit='us'))
        self.setattr_argument("repetitions", NumberValue(100, min=10, max=500, 
                                                         ndecimals=0))
        self.setattr_argument("nbins", NumberValue(100, ndecimals=0))
        self.setattr_argument("electrode", NumberValue(1, ndecimals=0,
                                                       scale=1, step=1))
        #  call fitting function
        self.fit = Rabi_Flopping(parent=self)

#==============================================================================
#  prepare stage for initialising datasets
#==============================================================================

    def prepare(self):
        self.xvar = "data.mw_3p3_2p2_t"
        self.xlabel = self.xvar + ".x"
        self.counts_label = self.xvar + ".counts"
        self.counts_hist_label = self.xvar + ".counts_hist"
        self.counts_hist_bins_label = self.xvar + ".counts_hist_bins"
        l = len(self.time_scan)
        #  set up size of arrays to hold data
        self.set_dataset(self.xlabel, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_label, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_label, np.empty((l, int(self.repetitions))),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_bins_label, np.linspace(0, 100, self.nbins+1),
                         broadcast=True, save=False)
        self.counts = [0]*int(self.repetitions)
[... 60 lines cut ...]

    @kernel
    def pulse_electrode(self, electrode, time):
        if electrode == 0:
            delay(time)
        elif electrode == 1:
            self.ttl_mw_ab.pulse(time)
        elif electrode == 2:
            with parallel:
                self.ttl_mw_mux.pulse(time)                
                self.ttl_mw_ab.pulse(time)
        elif electrode == 3:
            self.ttl_mw_mux.pulse(time)

#==============================================================================
#     analyse function to generate fit
#==============================================================================
    def analyze(self):
        tpi = self.fit.time_fit(self.xvar)
        if self.fit.save_fit:
            self.set_dataset("mw.tpi.3p3_2p2." + 'e' + str(self.electrode),
                             tpi, persist=True)
```

Where we look at the rabi flopping on the three different microwave electrodes we have. They have different pi times, so I wanted to the appropriate pi time to load depending on which electrode is chosen.

sbourdeauducq, 2016-06-29:
Why not have three arguments, one for each pi time?

r-srinivas, 2016-06-29:
We can also do that. Calling an argument from a widget before it doesn't work now, right?

sbourdeauducq, 2016-06-29:
When the repository is scanned, or when arguments are recomputed, all GUI argument values are ignored currently (and the defaults taken).

r-srinivas, 2016-06-29:
Okay. We can just work around this for now. It would be nice but not necessary.

closed 2016-06-29 by r-srinivas