m-labs/artiq #488: Same argument name with different groups

r-srinivas, 2016-06-24:
Using 1.1,

If two arguments have the same name but are in different groups, only the second one displays.

```python
from artiq.experiment import *

class Data_Label_test(EnvExperiment):

    def build(self):
        self.setattr_argument('test1', NumberValue(1), 'Category 1')
        self.setattr_argument('test1', NumberValue(self.test1), 'Category 2')

    def run(self):
        pass
```

The use case here is where we have a variable in the experiment and a value in the fit that would be nice if we could have the same name under different groups.

sbourdeauducq, 2016-06-27:
Groups are only decorative. How would you access ``test1`` in the experiment with the code above?

r-srinivas, 2016-06-28:
I should have picked a better example. The actual use would be more like this,

```from artiq.experiment import *

class Data_Label_test(EnvExperiment):

    def build(self):
        self.sc1 = Subcomponent1(parent=self)
        self.sc2 = Subcomponent2(parent=self)

    def run(self):
        pass

class Subcomponent1(HasEnvironment):

    def build(self):
        self.setattr_argument('test1', NumberValue(1), 'Subcomponent 1')

    def run(self):
        pass

class Subcomponent2(HasEnvironment):

    def build(self):
        self.setattr_argument('test1', NumberValue(1), 'Subcomponent 2')

    def run(self):
        pass
```

So you would call them differently anyway. One would be `self.sc1.test1', the other `self.sc2.test1`. However, only one of them shows up on the gui.

jordens, 2016-07-15:
Arguments are very much like datasets in the sense that their names need to be unique. Since you have to adhere to a global naming scheme for datasets already, is doing the same for arguments not intuitive and least confusing?

r-srinivas, 2016-07-15:
Sure, but the same way you could have a hierarchy in the datasets using `.`, it would have been nice to have similar structure with the arguments. The case I was thinking of was something along the lines of `self.tpi` from the experiment and `self.fit.tpi` from a fitting function instead of something like `self.fit.tpi_fit`.

jordens, 2016-07-15:
That would imply that from within a given subexperiment you can never access arguments used in sibling subexperiments or in parent experiments.

r-srinivas, 2016-07-15:
Can you do that currently?

```python
from artiq.experiment import *

class Data_Label_test(EnvExperiment):

    def build(self):
        self.sc1 = Subcomponent1(parent=self)
        self.sc2 = Subcomponent2(parent=self)
        self.setattr_argument('test0', NumberValue(0), 'Parent')

    def run(self):
        print(self.sc1.test1)
        print(self.sc2.test2)

class Subcomponent1(HasEnvironment):

    def build(self):
        self.setattr_argument('test1', NumberValue(1), 'Subcomponent 1')

class Subcomponent2(HasEnvironment):

    def build(self):
        self.setattr_argument('test2', NumberValue(2), 'Subcomponent 2')

```

From each subcomponent it seems like I can only access variables within it. If I try accessing `self.test0` from either of the subcomponents I get the error that those subcomponents don't have any attribute `test0`.

sbourdeauducq, 2016-08-05:
@r-srinivas Yes and it works fine here (release-1):

```
$ artiq_run.py test.py 
1.0
2.0
```

Please post the exact error message and steps to reproduce.

sbourdeauducq, 2016-08-05:
If you want ``test0`` in the subcomponents, you have to request it via ``get_argument``/``setattr_argument``, or have the parent pass it to them.

r-srinivas, 2016-08-08:
Okay. But any two subcomponents still can't share the same argument name, right?

closed 2016-08-08 by r-srinivas

r-srinivas, 2016-09-27:
I ran into a problem where I wanted to run the following experiment,

```python
class Raman_DSB(EnvExperiment):

    def build(self):
        self.rsb = Raman_SB_Scan(self, red=True)
        self.bsb = Raman_SB_Scan(self)

    def prepare(self):
        self.rsb.prepare()
        self.bsb.prepare()

    def run(self):
        self.rsb.run()
        self.bsb.run()

    def analyze(self):
        rsbfit = cf.fit_data(self.get_dataset(self.rsb.xlabel),
                              self.get_dataset(self.rsb.counts_label),
                              cf.AtomLine,
                              T=self.rsb.t_pulse)
        bsbfit = cf.fit_data(self.get_dataset(self.bsb.xlabel),
                              self.get_dataset(self.bsb.counts_label),
                              cf.AtomLine,
                              T=self.bsb.t_pulse)
        self.mutate_dataset(self.rsb.fitline_label,
                            (0, self.rsb.l),
                            rsbfit['fitline'])
        self.mutate_dataset(self.bsb.fitline_label,
                            (0, self.bsb.l),
                            bsbfit['fitline'])                          
        print(rsbfit['A'], bsbfit['A'])
```
The experiment runs as it should, one after the other but on the gui I only see the arguments from one of the two `Raman_SB_Scans`. The build looks like,

```python
class Raman_SB_Scan(Scan_1D):

    def _build(self, red=False):
        #  build hardware for experiment
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_ram1")
        self.setattr_device("ttl_ram2")
        self.setattr_device("ttl_ram1_ne")
        self.setattr_device("ttl_ram2_ne")
        self.setattr_device("core_dds")
        self.setattr_device("dds_ram1")
        self.setattr_device("dds_ram2")
        self.modes = ["axial", "radial_1", "radial_2",
                      "stretch", "rocking_1", "rocking_2"]
        self.frequencies = [self.get_dataset("raman."+m+'.f')
                            for m in self.modes]
        self.t_pis = [self.get_dataset("raman."+m+'.t_pi')
                      for m in self.modes]
        self.t_pi0s = [self.get_dataset("raman."+m+'.t_pi0')
                       for m in self.modes]
        self.mode_list = []
        self.red = red
        if self.red:
            label = "RSB"
        else:
            label = "BSB"
        for m in self.modes:
            i = self.modes.index(m)
            self.mode_list.append(
            ("mode: {mode}, " +
             "frequency: {frequency} MHz, " +
             "t_pi: {t_pi} us" +
             "t_pi0: {t_pi0} us").format(
             mode=m,
             frequency=str(round(self.frequencies[i]/1e6, 2)),
             t_pi=(str(round(self.t_pis[i]/1e-6, 2))),
             t_pi0=(str(round(self.t_pi0s[i]/1e-6, 2)))))
        self.setattr_argument("mode", EnumerationValue(self.mode_list),
                              label)
        self.setattr_argument('offset',
                              Scannable(default=[LinearScan(-0.1e6, 0.1e6, 100),
                                                 NoScan(0)],
                                        scale=1e6,
                                        unit='MHz',
                                        ndecimals=3),
                                        label)
        self.setattr_argument('flops', NumberValue(0.5, scale=1, ndecimals=2),
                              label)
        self.mw = Microwaves(self)
        self.cool = Cooling(self)
        print(self.red)
```

With `Scan_1D` calling `_build` in its `build` function. We can work around this by having two separate experiments, but there could be other experiments where this structure is useful. Having the arguments under different instances not showing up is problematic for that.

sbourdeauducq, 2016-09-28:
You could write a sub-experiment method such as:
```python
def setattr_argument_prefix(self, name, *args, **kwargs):
    setattr(self, name, self.get_argument(self.arg_prefix + name, *args, **kwargs)))
```
and set different values of ``self.arg_prefix`` in the sub-experiments.