m-labs/artiq #494: Error when round used with integers in kernel

r-srinivas, 2016-06-29:
Using 1.1 on Windows,

```python
from artiq.experiment import *

class Int_Test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        
    @kernel
    def run(self):
        print(round(1.0))
```

Gives me 1. 

```python
from artiq.experiment import *

class Int_Test(EnvExperiment):
    
    def build(self):
        self.setattr_device("core")
        
    @kernel
    def run(self):
        print(round(1))
```

Raises the following error,

```
ERROR:worker(13571,int_test.py):root:While compiling <repository>\test\code\int_
test.py
<repository>\test\code\int_test.py:36:21-36:22: error: cannot unify int(width='a
) with float
        print(round(1))
                    ^
<repository>\test\code\int_test.py:36:21-36:22: note: expression of type int(wid
th='a)
        print(round(1))
                    ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting
RID 13571
```

whitequark, 2016-06-29:
This is expected behavior. Rounding an integer is a no-op, and since kernel code has no polymorphism, such a round call is completely pointless as it will never perform any operation.

closed 2016-06-29 by whitequark

r-srinivas, 2016-06-29:
How is this done in regular python where `round(1)` just returns 1?

whitequark, 2016-06-29:
Regular Python has polymorphism; expressions can have many types. In ARTIQ Python any expression belongs to exactly one type, so if it is `int` then the `round` call can simply be removed.