m-labs/artiq #495: Chaining SPI transfers

r-srinivas, 2016-06-29:
Using 1.1, we wanted to communicate with some AD9914 using SPI. For the AD9914 driver, we have

```python
    @kernel
    def setup_bus(self, write_div=2, read_div=50):
        """Configure the SPI bus and the SPI transaction parameters
        for this device. This method has to be called before any other method
        if the bus has been used to access a different device in the meantime.

        This method advances the timeline by the duration of two
        RTIO-to-Wishbone bus transactions (6 coarse RTIO periods).

        :param write_div: Write clock divider.
        :param read_div: Read clock divider.
        """
        # write: 2*10ns >= 20ns = t_6 (clk falling to cs_n rising)
        # read: 50*10*ns >= 78ns = t_DV (clk falling to miso valid)
        self.bus.set_config_mu(_AD9914_SPI_CONFIG, write_div, read_div)
        # SPI core chip select is LSB of DDS channel
        self.bus.set_xfer(self.channel & 0x1, 20, 0)

    @kernel
    def write(self, address, data):
        """Write 40 bits of data (8 address bits, 32 data bits)

        This method advances the timeline by the duration of the SPI transfer.
        It also handles the appropriate chip selects.  

        Because two writes are involved, need to subtract one write_period_mu to
        rewind timeline from SPIMaster.write(), i.e. no cs deassertion between
        writes.
        """
        if self.channel & 0x2:
            self.cs2.on()
        else:
            self.cs2.off()
        #  can only write max 32 bits at a time
        write1 = (address << 12 | data >> 20)
        write2 = (data & 0xfffff)
        #  need to shift by 12 bits to the left
        #  write first 20 bits
        self.bus.write(write1 << 12)
        #  write next 20 bits
        self.bus.write(write2 << 12)
```

If I try to run just using this,

```python
    @kernel    
    def run(self):
        self.dds0.setup_bus()
        self.dds0.reset()
        self.dds0.write(0x01, 0x01010101)
```

On the scope you can see 2 sets of 20 bits being written with about 30 ns between them. 

[image]

As suggested in the spi.py file, I tried adding a negative delay to chain the writes. I added a delay of `delay_mu(-self.bus.write_period_mu)` which is `16 mu`, corresponding to 20 ns as we're using a 100 MHz clock. 

 ```python
    @kernel
    def write(self, address, data):
        """Write 40 bits of data (8 address bits, 32 data bits)

        This method advances the timeline by the duration of the SPI transfer.
        It also handles the appropriate chip selects.  

[... 325 lines cut ...]
#############################

#    @kernel
#    def read_channel_sync(self, channel=0, op=_AD5360_READ_X1A):
#        """Read a channel register.
#
#        This method advances the timeline by the duration of :meth:`write` plus
#        three RTIO-to-Wishbone transactions.
#
#        :param channel: Channel number to read from.
#        :param op: Operation to perform, one of :const:`_AD5360_READ_X1A`,
#          :const:`_AD5360_READ_X1B`, :const:`_AD5360_READ_OFFSET`,
#          :const:`_AD5360_READ_GAIN` (default: :const:`_AD5360_READ_X1A`).
#        :return: The 16 bit register value.
#        """
#        channel &= 0x3f
#        self.write(_AD5360_CMD_SPECIAL | _AD5360_SPECIAL_READ | op |
#                   _AD5360_READ_CHANNEL(channel))
#        self.bus.set_xfer(self.chip_select, 0, 24)
#        self.write(_AD5360_CMD_SPECIAL | _AD5360_SPECIAL_NOP)
#        self.bus.read_async()
#        self.bus.set_xfer(self.chip_select, 24, 0)
#        return self.bus.input_async() & 0xffff

```

jordens, 2016-06-30:
Could you try the following as your write sequence:

```python
self.bus.write(write1 << 12)
delay_mu(
    -(self.bus.xfer_period_mu + self.write_period_mu)  # undo bus.write() delay
    2*self.bus.ref_period_mu  # time to get the data through the double buffer into the shift register
)
self.bus.write(write2 << 12)
```

This is supposed to work according to my notes (It's also equivalent to `delay(-self.xfer_period_mu)` in this case but that doesn't matter)

r-srinivas, 2016-06-30:
Did you mean,

```python
        self.bus.write(write1 << 12)
        delay_mu(
            -(self.bus.xfer_period_mu + self.bus.write_period_mu) +  # undo bus.write() delay
            2*self.bus.ref_period_mu  # time to get the data through the double buffer into the shift register
        )
        self.bus.write(write2 << 12)
```
?

The sign at the end doesn't seem to matter. I've changed the bits being written to `       self.dds0.write(0x03, 0x03010101)` so it's clearer which sets of bits are being written. The second set of 20 bits aren't written at all.

[image]

Playing around with the delays, `self.bus.xfer_period_mu` is `20*self.bus.write_period_mu` which is `-320 mu`. `self.bus.write_period_mu` is `16 mu` and `self.bus.ref_period_mu` is `8 mu` for clarification.

For a delay of `-8 mu` or greater, the signal on the scope is,

[image]

For a delay less than '-8 mu', down to about '-320 mu', (in this case 10 but I've tried up to about -310 or so)

[image]

You have 40 clock cycles but lose the second set of 20 bits written.

And for `-320 mu` and below there only seem to be 20 clock cycles.

[image]

closed 2016-07-04 by jordens

jordens, 2016-07-04:
I also added more documentation for chaining transfers. Any (additional to what `bus.write()` applies) delay between `-ref_period_mu` and `- self.bus.xfer_period_mu - self.bus.write_period_mu + 3*self.bus.ref_period_mu` (both inclusive) should do it.

r-srinivas, 2016-07-05:
It's only `gateware\spi.py` that has changed, right? I tried patching that and still get similar behaviour.

- `-ref_period_mu (-8) <=delay <= 0` result in a delay between writes 
- `- self.bus.xfer_period_mu - self.bus.write_period_mu + 3*self.bus.ref_period_mu (-320-16+24=-312) <= delay < - ref_period_mu` results in the second 20 bits not being written but with 40 clock cycles
- and delays below that result in only 20 clock cycles. If it's low enough the delay results in a sequence error

jordens, 2016-07-05:
Did you flash that gateware?

r-srinivas, 2016-07-07:
That was it, oops. Seems to work now, thanks!

r-srinivas, 2016-07-07:
In the documentation that you added, what does the `2*self.bus.ref_period_mu` refer to? I've found that a delay more negative than `-2*self.bus.ref_period_mu` seems to result in timing errors for anything that happens subsequently. For example, if you have an IO update pulse after a couple of writes it happens during the write sequence. Keeping the delay at `-2*self.bus.ref_period_mu` seems to prevent that from happening.

jordens, 2016-07-07:
I can only see that happening if you are generally low on slack. Do you have code and/or an artiq_coreanalyzer trace?

If "a couple of writes" refers to more than two, then is what you saying equivalent to `chained transfers can happen every xfer_period_mu`?

r-srinivas, 2016-07-07:
Possibly, it depends what you mean can happen. The bits being written seems correct but the timing of the io_update pulse happens while the bits are being written.

jordens, 2016-07-08:
No. Being low on slack would result in an exception.
You have to look at how the xfers are scheduled and what happens to the timeline with all the delays that are applied to understand why that pulse happens during an xfer.

commit 71921de5 (2016-07-04): spi: do not shift when starting a xfer, closes #495
commit cda20ab2 (2016-07-04): spi: do not shift when starting a xfer, closes #495
commit cf364571 (2016-07-06): Revert "spi: do not shift when starting a xfer, closes #495"

This reverts commit 71921de5bde602b7e65f4a79dffda6a60f3ffe84.

It appears to prevent the comms CPU from booting. I do not
understand why.