m-labs/artiq #497: one alias for several devices

dkienzler, 2016-07-01:
is it possible to make a single alias in the device_database for e.g. multiple ttl channels?

sbourdeauducq, 2016-07-01:
How is that supposed to work? Toggle all TTLs at the same time?

sbourdeauducq, 2016-07-01:
And what about inputs?

dkienzler, 2016-07-01:
yes, that was the idea.
We often have the situation that we need to turn many TTLs on at the same time.
In older systems that was often solved with a bit mask, so you could just add up the ttl channels and toggle them together. I see that that's not possible here, but it would still be nice to have something similar on the low level. 
if i want to do this right now i need to do a setattr_device for each physical channel and then a pulse for each of them, with parallel. is that correct?

we used to do this with inputs and outputs. I don't know how to deal with that in the current framework.

sbourdeauducq, 2016-07-01:
You can create a class that contains methods that toggle the multiple TTLs. Something like:
```python
@kernel
def on(self):
   self.ttl1.on()
   self.ttl2.on()
```

or:
```python
@kernel
def on(self):
   for ttl in self.ttls:
       ttl.on()
```

How is an input from several TTLs supposed to work?

closed 2016-07-02 by sbourdeauducq