m-labs/artiq #500: Missing pulses for multiple runs of subclass

cjbe, 2016-07-06:
I have an experiment, BugScan, that runs a subclass BugSubRoutine multiple times. On the first run of BugSunRoutine the RTIO outputs are as expected, but for subsequent runs the RTIO events generated by bug_package.profileSwitcher are missing:
[image]
I would expect the three executions of self.sub_routine.run() to generate identical results.

I am using Artiq 1.1, with the nist_clock gateware on a KC705.

Experiment code:
```python
from artiq.experiment import *

class BugScan(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.sub_routine = BugSubRoutine(*self.managers())
        
    def run(self):
        for i in range(3):
            print(i)
            self.sub_routine.run()

class BugSubRoutine(HasEnvironment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("bugdev")
        self.setattr_device("ttl0")

    @kernel
    def run_scan_point(self, j):
        self.core.break_realtime()
        for i in range(10):
            delay(1*ms)
            self.bugdev.set_profile(0)
            delay(1*ms)
            if j != 0:
                self.bugdev.set_profile(j)
                delay(10*us)
                self.ttl0.pulse(1*ms)

    @kernel
    def run_scan(self):
        for j in range(3):
            self.run_scan_point(j)

    def run(self):
        self.run_scan()
```

Mediator class, bug_package.profileSwitcher:
```python
from artiq.language.core import *

class ProfileSwitcher:
    def __init__(self, dmgr, profile_lines):
        self.n_profiles = len(profile_lines)
        self.profiles = []
        for p in profile_lines:
            self.profiles.append(dmgr.get(p))

    @kernel
    def set_profile(self, profile):
        for i in range(self.n_profiles):
            if profile & (2**i):
                self.profiles[i].on()
            else:
                self.profiles[i].off()
```

Device entry in device_db:
```
"bugdev": {
        "type": "local",
        "module": "bug_package.profileSwitcher",
        "class": "ProfileSwitcher",
        "arguments": {
            "profile_lines": ["ttl4", "ttl5", "ttl6"]
        }
    }
```

sbourdeauducq, 2016-07-06:
Can you trim this down a bit?

cjbe, 2016-07-06:
I have managed to trim it down substaintionally - the problem can be triggered without using a subclass. I had not noticed this initially as the bug appears and dissapears with small changes in the code.

The following experiment, along with the mediator class in my earlier post, triggers the bug.
[image]
```python
from artiq.experiment import *

class BugScan(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("bugdev")
        self.setattr_device("ttl0")

    @kernel
    def kernel_func(self):
        self.core.break_realtime()
        self.bugdev.set_profile(0)
        delay(1*ms)
        self.bugdev.set_profile(1)
        self.ttl0.pulse(10*ms)

    def run(self):
        for i in range(3):
            print(i)
            self.kernel_func()
```

sbourdeauducq, 2016-07-06:
Are the diagrams made with artiq_coreanalyzer? Could it be some experiment switching related issue - what happens if you do ``time.sleep(1)`` at the end of ``run``?

sbourdeauducq, 2016-07-06:
And can you keep removing code? There's still a lot.

cjbe, 2016-07-06:
The diagrams are made from artiq_coreanalyzer, and match the actual TTL outputs.
Adding a sleep(1) after the call to kernel_func() does not fix the bug.

cjbe, 2016-07-06:
I have managed to simplify it further - the experiment is now self-contained.

```python
from artiq.experiment import *

class BugScan(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")

        self.profiles = [self.get_device(dev) for dev in ["ttl4", "ttl5"]]
        self.n_profiles = len(self.profiles)

    @kernel
    def set_profile(self, profile):
        for i in range(self.n_profiles):
            if profile & (2**i):
                self.profiles[i].on()
            else:
                self.profiles[i].off()
            delay(20*ms)

    @kernel
    def kernel_func(self):
        self.core.break_realtime()
        self.set_profile(0)
        delay(50*ms)
        self.set_profile(1)
        self.ttl0.pulse(50*ms)

    @kernel
    def kernel_test(self):
        self.core.break_realtime()
        for i in range(4):
            self.set_profile(i)
            delay(50*ms)

    def run(self):
        self.kernel_test()
        for i in range(4):
            print(i)
            self.kernel_func()
```
[image]

The kernel_test() function runs set_profile for i=0..3, this is correctly output on to [ttl4,ttl5].
Then kernel_func() calls set_profile(0) and set_profile(1), but the profile lines do not change properly, and it seems like the self.profiles[i].on() / off() calls set ttl5 for i=0 and i=1.

sbourdeauducq, 2016-07-07:
Reproduced. More trimming (symptom is TTL5 is high at the end):
```python
from artiq.experiment import *

class BugScan(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")

        self.profiles = [self.get_device(dev) for dev in ["ttl4", "ttl5"]]
        self.n_profiles = len(self.profiles)

    @kernel
    def set_profile(self, profile):
        for i in range(self.n_profiles):
            if profile & (2**i):
                self.profiles[i].on()
            else:
                self.profiles[i].off()

    @kernel
    def kernel_test(self):
        self.core.break_realtime()
        self.set_profile(3)

    @kernel
    def kernel_func(self):
        self.core.break_realtime()
        self.set_profile(0)
        delay(50*ms)
        self.set_profile(1)
        self.ttl0.pulse(50*ms)

    def run(self):
        self.kernel_test()
        self.kernel_func()
```

sbourdeauducq, 2016-07-07:
@whitequark The compiler's attribute writeback feature corrupts the profile list.

```python
from artiq.experiment import *

class BugScan(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.profiles = [self.get_device(dev) for dev in ["ttl4", "ttl5"]]
        self.n_profiles = len(self.profiles)

    @kernel
    def set_profile(self, profile):
        for i in range(self.n_profiles):
            if profile & (2**i):
                self.profiles[i].on()
            else:
                self.profiles[i].off()

    @kernel
    def kernel_test(self):
        self.core.break_realtime()
        self.set_profile(3)

    def run(self):
        print(self.profiles)
        self.kernel_test()
        print(self.profiles)
```

```
$ artiq_run.py bugscan.py 
[<artiq.coredevice.ttl.TTLOut object at 0x7eff5f094748>, <artiq.coredevice.ttl.TTLOut object at 0x7eff5f094b38>]
[<artiq.coredevice.ttl.TTLOut object at 0x7eff5f094748>, <artiq.coredevice.ttl.TTLOut object at 0x7eff5f094748>]
```

Note how the list incorrectly contains twice the first object after the kernel has been run.

sbourdeauducq, 2016-07-07:
If that's not straightforward to fix, I propose dropping attribute writeback. It's not documented and I doubt anyone uses it, and dataset management has developed since then.

whitequark, 2016-07-07:
This has nothing to do with attribute writeback. Rather, any list `[a, b, c...]` which contains objects of anything but builtin types will arrive as `[a, a, a...]`.

commit 653eeb47 (2016-07-07): runtime: fix serialization of object lists.

Fixes #500.
commit 6a1706b8 (2016-07-07): runtime: fix serialization of object lists.

Fixes #500.
closed 2016-07-07 by whitequark