m-labs/artiq #502: Rule violation when trying to generate bitstream for qc2

r-srinivas, 2016-07-06:
I was trying to generate the bitstream for the release-1 branch (commit 07b41763c20ac7470d723d8e30e937adb0250ac9) and ran into the following error,

```
ERROR: [DRC 23-20] Rule violation (BIVC-1) Bank IO standard Vcc - Conflicting Vcc voltages in bank 12. For example, the following two ports in this bank have conflicting VCCOs:  
user_sma_gpio_n (LVCMOS25, requiring VCCO=2.500) and dds_d[0] (LVTTL, requiring VCCO=3.300)
```

This results in an oserror later,

```
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

    while executing
"place_design"
    (file "top.tcl" line 99)
```

That was from `vivado.log`. On the terminal I see,

```
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

    while executing
"place_design"
    (file "top.tcl" line 99)
INFO: [Common 17-206] Exiting Vivado at Wed Jul  6 16:25:06 2016...
Traceback (most recent call last):
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/rabi/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rabi/artiq-dev-1.1/artiq/artiq/gateware/targets/kc705.py", line 408, in <module>
    main()
  File "/home/rabi/artiq-dev-1.1/artiq/artiq/gateware/targets/kc705.py", line 404, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/rabi/artiq-dev-1.1/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/rabi/artiq-dev-1.1/misoc/misoc/integration/builder.py", line 156, in build
    run=self.compile_gateware, **kwargs)
  File "/home/rabi/artiq-dev-1.1/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev-1.1/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/rabi/artiq-dev-1.1/migen/migen/build/xilinx/vivado.py", line 131, in build
    _run_vivado(build_name, toolchain_path, source)
  File "/home/rabi/artiq-dev-1.1/migen/migen/build/xilinx/vivado.py", line 70, in _run_vivado
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

How can I fix this?

r-srinivas, 2016-07-06:
Full vivado.log

```
#-----------------------------------------------------------
# Vivado v2015.4 (64-bit)
# SW Build 1412921 on Wed Nov 18 09:44:32 MST 2015
# IP Build 1412160 on Tue Nov 17 13:47:24 MST 2015
# Start of session at: Wed Jul  6 15:43:04 2016
# Process ID: 7331
# Current directory: /home/rabi/artiq-dev-1.1/misoc_nist_qc2_kc705/gateware
# Command line: vivado -mode batch -source top.tcl
# Log file: /home/rabi/artiq-dev-1.1/misoc_nist_qc2_kc705/gateware/vivado.log
# Journal file: /home/rabi/artiq-dev-1.1/misoc_nist_qc2_kc705/gateware/vivado.jou
#-----------------------------------------------------------
source top.tcl
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_simple_dpram_sclk.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_simple_dpram_sclk.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_simple_dpram_sclk.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cpu_cappuccino.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cpu_cappuccino.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cpu_cappuccino.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_bus_if_avalon.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_bus_if_avalon.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_bus_if_avalon.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx-sprs.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx-sprs.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx-sprs.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_dmmu.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_dmmu.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_dmmu.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_lsu_espresso.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_lsu_espresso.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_lsu_espresso.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_muldiv.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_muldiv.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/pfpu32/pfpu32_muldiv.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_execute_alu.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_execute_alu.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_execute_alu.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cache_lru.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cache_lru.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_cache_lru.v:]
# add_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_fetch_espresso.v}
# set_property library work [get_files {/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_fetch_espresso.v}]
WARNING: [filemgmt 56-99] Vivado Synthesis ignores library specification for Verilog or SystemVerilog files. [/home/rabi/artiq-dev-1.1/misoc/misoc/cores/mor1kx/verilog/rtl/verilog/mor1kx_fetch_espresso.v:]
[... cut ...]
	Parameter IS_D8_INVERTED bound to: 1'b0 
	Parameter IS_T1_INVERTED bound to: 1'b0 
	Parameter IS_T2_INVERTED bound to: 1'b0 
	Parameter IS_T3_INVERTED bound to: 1'b0 
	Parameter IS_T4_INVERTED bound to: 1'b0 
	Parameter SERDES_MODE bound to: MASTER - type: string 
	Parameter SRVAL_OQ bound to: 1'b0 
	Parameter SRVAL_TQ bound to: 1'b0 
	Parameter TBYTE_CTL bound to: FALSE - type: string 
	Parameter TBYTE_SRC bound to: FALSE - type: string 
	Parameter TRISTATE_WIDTH bound to: 1 - type: integer 
```

r-srinivas, 2016-07-06:
```
INFO: [Synth 8-256] done synthesizing module 'OSERDESE2__parameterized25' (33#1) [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:25860]
INFO: [Synth 8-638] synthesizing module 'ISERDESE2__parameterized26' [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:19667]
	Parameter DATA_RATE bound to: DDR - type: string 
	Parameter DATA_WIDTH bound to: 8 - type: integer 
	Parameter DYN_CLKDIV_INV_EN bound to: FALSE - type: string 
	Parameter DYN_CLK_INV_EN bound to: FALSE - type: string 
	Parameter INIT_Q1 bound to: 1'b0 
	Parameter INIT_Q2 bound to: 1'b0 
	Parameter INIT_Q3 bound to: 1'b0 
	Parameter INIT_Q4 bound to: 1'b0 
	Parameter INTERFACE_TYPE bound to: NETWORKING - type: string 
	Parameter IOBDELAY bound to: NONE - type: string 
	Parameter IS_CLKB_INVERTED bound to: 1'b0 
	Parameter IS_CLKDIVP_INVERTED bound to: 1'b0 
	Parameter IS_CLKDIV_INVERTED bound to: 1'b0 
	Parameter IS_CLK_INVERTED bound to: 1'b0 
	Parameter IS_D_INVERTED bound to: 1'b0 
	Parameter IS_OCLKB_INVERTED bound to: 1'b0 
	Parameter IS_OCLK_INVERTED bound to: 1'b0 
	Parameter NUM_CE bound to: 1 - type: integer 
	Parameter OFB_USED bound to: FALSE - type: string 
	Parameter SERDES_MODE bound to: MASTER - type: string 
	Parameter SRVAL_Q1 bound to: 1'b0 
	Parameter SRVAL_Q2 bound to: 1'b0 
	Parameter SRVAL_Q3 bound to: 1'b0 
	Parameter SRVAL_Q4 bound to: 1'b0 
INFO: [Synth 8-256] done synthesizing module 'ISERDESE2__parameterized26' (33#1) [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:19667]
INFO: [Synth 8-638] synthesizing module 'OSERDESE2__parameterized26' [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:25860]
	Parameter DATA_RATE_OQ bound to: DDR - type: string 
	Parameter DATA_RATE_TQ bound to: BUF - type: string 
	Parameter DATA_WIDTH bound to: 8 - type: integer 
	Parameter INIT_OQ bound to: 1'b0 
	Parameter INIT_TQ bound to: 1'b0 
	Parameter IS_CLKDIV_INVERTED bound to: 1'b0 
	Parameter IS_CLK_INVERTED bound to: 1'b0 
	Parameter IS_D1_INVERTED bound to: 1'b0 
	Parameter IS_D2_INVERTED bound to: 1'b0 
	Parameter IS_D3_INVERTED bound to: 1'b0 
	Parameter IS_D4_INVERTED bound to: 1'b0 
	Parameter IS_D5_INVERTED bound to: 1'b0 
	Parameter IS_D6_INVERTED bound to: 1'b0 
	Parameter IS_D7_INVERTED bound to: 1'b0 
	Parameter IS_D8_INVERTED bound to: 1'b0 
	Parameter IS_T1_INVERTED bound to: 1'b0 
	Parameter IS_T2_INVERTED bound to: 1'b0 
	Parameter IS_T3_INVERTED bound to: 1'b0 
	Parameter IS_T4_INVERTED bound to: 1'b0 
	Parameter SERDES_MODE bound to: MASTER - type: string 
	Parameter SRVAL_OQ bound to: 1'b0 
	Parameter SRVAL_TQ bound to: 1'b0 
	Parameter TBYTE_CTL bound to: FALSE - type: string 
	Parameter TBYTE_SRC bound to: FALSE - type: string 
	Parameter TRISTATE_WIDTH bound to: 1 - type: integer 
INFO: [Synth 8-256] done synthesizing module 'OSERDESE2__parameterized26' (33#1) [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:25860]
INFO: [Synth 8-638] synthesizing module 'ISERDESE2__parameterized27' [/opt/Xilinx/Vivado/2015.4/scripts/rt/data/unisim_comp.v:19667]
	Parameter DATA_RATE bound to: DDR - type: string 
	Parameter DATA_WIDTH bound to: 8 - type: integer 
	Parameter DYN_CLKDIV_INV_EN bound to: FALSE - type: string 
	Parameter DYN_CLK_INV_EN bound to: FALSE - type: string 
	Parameter INIT_Q1 bound to: 1'b0 
	Parameter INIT_Q2 bound to: 1'b0 
	Parameter INIT_Q3 bound to: 1'b0 
	Parameter INIT_Q4 bound to: 1'b0 
	Parameter INTERFACE_TYPE bound to: NETWORKING - type: string 
	Parameter IOBDELAY bound to: NONE - type: string 
	Parameter IS_CLKB_INVERTED bound to: 1'b0 
	Parameter IS_CLKDIVP_INVERTED bound to: 1'b0 
	Parameter IS_CLKDIV_INVERTED bound to: 1'b0 
	Parameter IS_CLK_INVERTED bound to: 1'b0 
[... 2195 lines cut ...]
Report RTL Partitions: 
+------+--------------+------------+----------+
|      |RTL Partition |Replication |Instances |
+------+--------------+------------+----------+
|1     |top__GCB0     |           1|     37258|
|2     |top__GCB1     |           1|     12564|
|3     |top__GCB2     |           1|     10785|
|4     |top__GCB3     |           1|     18001|
|5     |top__GCB4     |           1|     20574|
|6     |top__GCB5     |           1|     28261|
|7     |top__GCB6     |           1|     10659|
|8     |top__GCB7     |           1|     42848|
|9     |top__GCB8     |           1|      9347|
+------+--------------+------------+----------+
INFO: [Synth 8-4652] Swapped enable and write-enable on 1 RAM instances of RAM icache_gen.mor1kx_icache/way_memories[0].way_data_ram/mem_reg to conserve power
Optimized 2 bits of RAM "store_buffer_gen.mor1kx_store_buffer/fifo_ram/mem_reg" due to constant propagation. Old ram width 101 bits, new ram width 99 bits
INFO: [Synth 8-4652] Swapped enable and write-enable on 1 RAM instances of RAM dcache_gen.mor1kx_dcache/way_memories[0].way_data_ram/mem_reg to conserve power
Optimized 4 bits of RAM "storage_4_reg" due to constant propagation. Old ram width 41 bits, new ram width 37 bits
WARNING: [Synth 8-115] binding instance 'i_1' in module 'extram__243' to reference 'logic__5275' which has no pins
INFO: [Synth 8-4652] Swapped enable and write-enable on 1 RAM instances of RAM icache_gen.mor1kx_icache/way_memories[0].way_data_ram/mem_reg to conserve power
Optimized 2 bits of RAM "store_buffer_gen.mor1kx_store_buffer/fifo_ram/mem_reg" due to constant propagation. Old ram width 101 bits, new ram width 99 bits
INFO: [Synth 8-4652] Swapped enable and write-enable on 1 RAM instances of RAM dcache_gen.mor1kx_dcache/way_memories[0].way_data_ram/mem_reg to conserve power
Optimized 8 bits of RAM "tag_mem_reg" due to constant propagation. Old ram width 24 bits, new ram width 16 bits

```

r-srinivas, 2016-07-06:
```
---------------------------------------------------------------------------------
Start ROM, RAM, DSP and Shift Register Reporting
---------------------------------------------------------------------------------

Block RAM:
+------------+---------------------------------------------------------------+------------------------+---+---+------------------------+---+---+--------------+--------+--------+-----------------------------------------------------------+
|Module Name | RTL Object                                                    | PORT A (Depth x Width) | W | R | PORT B (Depth x Width) | W | R | OUT_REG      | RAMB18 | RAMB36 | Hierarchical Name                                         | 
+------------+---------------------------------------------------------------+------------------------+---+---+------------------------+---+---+--------------+--------+--------+-----------------------------------------------------------+
|top__GCB0   | icache_gen.mor1kx_icache/tag_ram/mem_reg                      | 256 x 20(READ_FIRST)   | W |   | 256 x 20(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | mor1kx_cpu_cappuccino/mor1kx_fetch_cappuccino/extram__199 | 
|top__GCB0   | icache_gen.mor1kx_icache/way_memories[0].way_data_ram/mem_reg | 1 K x 32(READ_FIRST)   | W |   | 1 K x 32(WRITE_FIRST)  |   | R | Port A and B | 0      | 1      | mor1kx_cpu_cappuccino/mor1kx_fetch_cappuccino/extram__201 | 
|top__GCB0   | store_buffer_gen.mor1kx_store_buffer/fifo_ram/mem_reg         | 256 x 101(READ_FIRST)  | W |   | 256 x 101(WRITE_FIRST) |   | R | Port A and B | 1      | 1      | mor1kx_cpu_cappuccino/mor1kx_lsu_cappuccino/extram__203   | 
|top__GCB0   | dcache_gen.mor1kx_dcache/tag_ram/mem_reg                      | 256 x 20(READ_FIRST)   | W |   | 256 x 20(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | mor1kx_cpu_cappuccino/mor1kx_lsu_cappuccino/extram__205   | 
|top__GCB0   | dcache_gen.mor1kx_dcache/way_memories[0].way_data_ram/mem_reg | 1 K x 32(READ_FIRST)   | W |   | 1 K x 32(WRITE_FIRST)  |   | R | Port A and B | 0      | 1      | mor1kx_cpu_cappuccino/mor1kx_lsu_cappuccino/extram__207   | 
|top__GCB0   | rfa/mem_reg                                                   | 32 x 32(READ_FIRST)    | W |   | 32 x 32(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | mor1kx_cpu_cappuccino/mor1kx_rf_cappuccino/extram__209    | 
|top__GCB0   | rfb/mem_reg                                                   | 32 x 32(READ_FIRST)    | W |   | 32 x 32(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | mor1kx_cpu_cappuccino/mor1kx_rf_cappuccino/extram__211    | 
|top__GCB0   | storage_103_reg                                               | 128 x 32               | W |   | 128 x 32(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | extram__213                                               | 
|top__GCB0   | storage_101_reg                                               | 128 x 32               | W |   | 128 x 32(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | extram__215                                               | 
|top__GCB0   | storage_99_reg                                                | 128 x 32               | W |   | 128 x 32(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | extram__217                                               | 
|top__GCB0   | storage_97_reg                                                | 128 x 32               | W |   | 128 x 32(WRITE_FIRST)  |   | R | Port A and B | 1      | 0      | extram__219                                               | 
|top__GCB0   | mem_grain0_1_reg                                              | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__226                                               | 
|top__GCB0   | mem_grain1_1_reg                                              | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__228                                               | 
|top__GCB0   | mem_grain2_1_reg                                              | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__230                                               | 
|top__GCB0   | mem_grain3_1_reg                                              | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__232                                               | 
|top__GCB0   | mem_grain0_reg                                                | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__234                                               | 
|top__GCB0   | mem_grain1_reg                                                | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__236                                               | 
|top__GCB0   | mem_grain2_reg                                                | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__238                                               | 
|top__GCB0   | mem_grain3_reg                                                | 512 x 8(WRITE_FIRST)   | W | R | 512 x 8(WRITE_FIRST)   |   | R | Port A and B | 1      | 0      | extram__240                                               | 
[... cut ...]
INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
ERROR: [Vivado_Tcl 4-23] Error(s) found during DRC. Placer not run.
INFO: [Common 17-83] Releasing license: Implementation
8 Infos, 58 Warnings, 0 Critical Warnings and 2 Errors encountered.
place_design failed
ERROR: [Common 17-39] 'place_design' failed due to earlier errors.

    while executing
"place_design"
    (file "top.tcl" line 99)
INFO: [Common 17-206] Exiting Vivado at Wed Jul  6 15:46:54 2016...
```

sbourdeauducq, 2016-07-07:
This worked on the build server. Is your migen up to date?

r-srinivas, 2016-07-07:
I'm pretty sure I had as I had set up that release folder from scratch but
I'll check again tomorrow.

sbourdeauducq, 2016-07-07:
It works with the migen release but not with migen master. This commit makes it work with both.

closed 2016-07-07 by sbourdeauducq

r-srinivas, 2016-07-07:
That worked, thanks.

commit 8cb29fcb (2016-07-07): targets/kc705: redefine user SMAs as 3.3V IO. Closes #502
commit adcf53f1 (2016-07-07): targets/kc705: redefine user SMAs as 3.3V IO. Closes #502