m-labs/artiq #506: OSError (invalid argument) in test_start_ping_stop_controller test on windows

jordens, 2016-07-12:
Currently http://buildbot.m-labs.hk/builders/artiq-win64-test/builds/256/steps/python_unittest/logs/stdio

Initial failures: http://buildbot.m-labs.hk/builders/artiq-win64-test/builds/244/steps/python_unittest/logs/stdio

sbourdeauducq, 2016-07-12:
Not user-visible and only with Python 3.5.2, so not that serious.

whitequark, 2016-07-12:
Well, we need to fix the CI anyway, because there is no point in even having it if it's always red.

whitequark, 2016-07-12:
(Also, right now docs aren't updated at the end of the build, because win64 fails first.)

sbourdeauducq, 2016-07-12:
Yes. But it still runs the other tests and the same ones on Linux.

whitequark, 2016-07-12:
It is still useless because of how hard to notice it is that the kind of failure has changed.

jordens, 2016-07-12:
IMHO all that and if it is not serious then we don't need to hold 1.2.

sbourdeauducq, 2016-07-12:
http://bugs.python.org/issue27500

sbourdeauducq, 2016-07-12:
Workarounds: use ``127.0.0.1`` or ``localhost``.

cr1901, 2016-07-12:
Problem commit/line is: https://github.com/python/cpython/commit/03df54d549173e17e1cf9a767199de32a363aa6b#diff-08afa52ab2b1511bee8527814ad44d80R138

Python 3.5.2 tries to skip calling `getaddrinfo()`, and the function which does that (`_ipaddr_info()`) should be returning a 4-tuple for AF_INET6 family. Instead it only returns 2, and the Windows function `ConnectEx()` doesn't like that some IPv6 fields are missing.

commit 00ed3909 (2017-02-28): monkey-patch Python 3.5.3 asyncio.base_events._ipaddr_info
this fixes #506 for py3.5.3
commit 056a5a7e (2017-02-28): Merge branch 'py3.5.3_monkey-patch' of https://github.com/cjbe/artiq

* 'py3.5.3_monkey-patch' of https://github.com/cjbe/artiq:
  monkey-patch Python 3.5.3 asyncio.base_events._ipaddr_info this fixes #506 for py3.5.3
commit 349774e6 (2017-02-28): monkey-patch Python 3.5.3 asyncio.base_events._ipaddr_info
this fixes #506 for py3.5.3
closed 2016-07-13 by sbourdeauducq