m-labs/artiq #507: Dependencies missing in dev conda channel for windows

r-srinivas, 2016-07-12:
I tried installing artiq from the dev channel and ran into the following error,

```
c:\artiq-dev-master\artiq-magtrap>conda create -n artiq-dev-master artiq
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ......
Solving package specifications: .
Error:  Dependencies missing in current win-64 channels:
  - artiq -> pyelftools
  - artiq -> llvmlite-artiq 0.5.1 py35_17
  - artiq -> prettytable
  - artiq -> levenshtein
  - artiq -> pydaqmx
  - artiq -> binutils-or1k-linux
  - artiq -> llvmlite-artiq 0.5.1
  - artiq -> quamash
  - artiq -> llvmlite-artiq 0.10.0.dev py35_22
  - artiq -> llvmlite-artiq 0.10.0.dev py35_20
  - artiq -> lit
  - artiq -> outputcheck
  - artiq -> flterm
  - artiq -> pygit2
  - artiq -> readline
  - artiq -> pythonparser
  - artiq -> pyqt5
  - artiq -> sphinx-argparse
  - artiq -> llvmlite-artiq 0.10.0.dev py35_24
  - artiq -> aiohttp
  - artiq -> asyncserial

Did you mean one of these?

    llvmlite-artiq

Did you mean one of these?

    pycparser

Did you mean one of these?

    pykit

Did you mean one of these?

    pyqt

Did you mean one of these?

    pyreadline

Did you mean one of these?

    pyserial

Did you mean one of these?

    blist

Did you mean one of these?

    pytables

You can search for this package on anaconda.org with

    anaconda search -t conda outputcheck

 (and similarly for the other packages)
```

Looking at the channels using `conda info`

```
c:\artiq-dev-master\artiq-magtrap>conda info
Using Anaconda Cloud api site https://api.anaconda.org
Current conda install:

             platform : win-64
        conda version : 4.0.8
  conda-build version : 1.20.0
       python version : 3.5.1.final.0
     requests version : 2.9.1
     root environment : C:\Anaconda3  (writable)
  default environment : C:\Anaconda3
     envs directories : C:\Anaconda3\envs
        package cache : C:\Anaconda3\pkgs
         channel URLs : http://conda.anaconda.org/m-labs/label/dev/win-64/
                        http://conda.anaconda.org/m-labs/label/dev/noarch/
                        https://repo.continuum.io/pkgs/free/win-64/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/win-64/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : C:\Users\rabi\.condarc
    is foreign system : False

```

I could install it previously using `conda create -n artiq-dev-master -c http://conda.anaconda.org/m-labs/label/dev artiq` but it installed only artiq from the dev channel but its dependencies from the main channel. I then couldn't launch the artiq_dashboard. I thought that's maybe from some conflict in the dependent packages so tried to install it all from the dev channel.

whitequark, 2016-07-12:
You need to add both dev and main channels.

r-srinivas, 2016-07-12:
Okay, thanks. That's what I had done previously but I couldn't get the artiq_dashboard to work.