m-labs/artiq #508: Dashboard does not launch on windows

r-srinivas, 2016-07-12:
I installed artiq through conda from the dev channel on windows, and tried launching the dashboard. I run into the following error

```
[artiq-dev-master] c:\artiq-dev-master\artiq-magtrap>artiq_dashboard
Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq-dev-master\Scripts\artiq_dashboard-script.py", l
ine 8, in <module>
    load_entry_point('artiq==2.0.dev0+482.gc5ba44b', 'gui_scripts', 'artiq_dashb
oard')()
  File "c:\artiq-dev-master\artiq\artiq\frontend\artiq_dashboard.py", line 107,
in main
    args.server, args.port_control, "master_" + target))
  File "c:\anaconda3\envs\artiq-dev-master\lib\site-packages\quamash-0.5.5-py3.5
.egg\quamash\__init__.py", line 273, in run_until_complete
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\futures.py", line 274, in
 result
    raise self._exception
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\tasks.py", line 239, in _
step
    result = coro.send(None)
  File "c:\artiq-dev-master\artiq\artiq\protocols\pc_rpc.py", line 199, in conne
ct_rpc
    await asyncio.open_connection(host, port, limit=4*1024*1024)
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\streams.py", line 75, in
open_connection
    lambda: protocol, host, port, **kwds)
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\base_events.py", line 695
, in create_connection
    raise exceptions[0]
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\base_events.py", line 682
, in create_connection
    yield from self.sock_connect(sock, address)
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\proactor_events.py", line
 443, in sock_connect
    return self._proactor.connect(sock, address)
  File "c:\anaconda3\envs\artiq-dev-master\lib\site-packages\quamash-0.5.5-py3.5
.egg\quamash\_windows.py", line 135, in connect
  File "C:\Anaconda3\envs\artiq-dev-master\lib\asyncio\windows_events.py", line
506, in connect
    ov.ConnectEx(conn.fileno(), address)
OSError: [WinError 10022] An invalid argument was supplied
```

The packages I have installed in this environment are

```
[artiq-dev-master] c:\artiq-dev-master\artiq-magtrap>conda list
# packages in environment at C:\Anaconda3\envs\artiq-dev-master:
#
aiohttp                   0.17.2                     py_3    http://conda.anacon
da.org/m-labs/label/main/noarch/aiohttp-0.17.2-py_3.tar.bz2
alabaster                 0.7.8                    py35_0    defaults
artiq                     2.0.dev         py_482+gitc5ba44b    http://conda.anac
onda.org/m-labs/label/dev/noarch/artiq-2.0.dev-py_482+gitc5ba44b.tar.bz2
asyncserial               0.1             py_11+git758c1e5    http://conda.anaco
nda.org/m-labs/label/main/noarch/asyncserial-0.1-py_11+git758c1e5.tar.bz2
babel                     2.3.3                    py35_0    defaults
binutils-or1k-linux       2.25.1                        4    http://conda.anacon
da.org/m-labs/label/main/win-64/binutils-or1k-linux-2.25.1-4.tar.bz2
cffi                      1.6.0                    py35_0    defaults
chardet                   2.3.0                    py35_0    defaults
colorama                  0.3.7                    py35_0    defaults
dateutil                  2.4.2                      py_2    http://conda.anacon
da.org/m-labs/label/main/noarch/dateutil-2.4.2-py_2.tar.bz2
docutils                  0.12                     py35_2    defaults
h5py                      2.6.0               np111py35_1    defaults
hdf5                      1.8.16                   vc14_0  [vc14]  defaults
icu                       56.1                          0    http://conda.anacon
da.org/m-labs/label/main/win-64/icu-56.1-0.tar.bz2
imagesize                 0.7.1                    py35_0    defaults
jinja2                    2.8                      py35_1    defaults
jpeg                      8d                       vc14_0  [vc14]  defaults
levenshtein               0.12.0                   py35_1    http://conda.anacon
da.org/m-labs/label/main/win-64/levenshtein-0.12.0-py35_1.tar.bz2
libgit2                   0.24.1                        7    http://conda.anacon
da.org/m-labs/label/main/win-64/libgit2-0.24.1-7.tar.bz2
libpng                    1.6.22                   vc14_0  [vc14]  defaults
libssh2                   1.7.0.git                     5    http://conda.anacon
da.org/m-labs/label/main/win-64/libssh2-1.7.0.git-5.tar.bz2
lit                       0.4.1                      py_2    http://conda.anacon
da.org/m-labs/label/main/noarch/lit-0.4.1-py_2.tar.bz2
llvmlite-artiq            0.10.0.dev              py35_24    http://conda.anacon
da.org/m-labs/label/main/win-64/llvmlite-artiq-0.10.0.dev-py35_24.tar.bz2
markupsafe                0.23                     py35_2    defaults
misoc (c:\misoc\misoc)    0.1                       <pip>
mkl                       11.3.3                        1    defaults
numpy                     1.11.1                   py35_0    defaults
openssl                   1.0.2h                   vc14_0  [vc14]  defaults
outputcheck               0.4.2                      py_1    http://conda.anacon
da.org/m-labs/label/main/noarch/outputcheck-0.4.2-py_1.tar.bz2
pip                       8.1.2                    py35_0    defaults
prettytable               0.7.2                      py_3    http://conda.anacon
da.org/m-labs/label/main/noarch/prettytable-0.7.2-py_3.tar.bz2
pycparser                 2.14                     py35_1    defaults
pygit2                    0.24.0                   py35_4    http://conda.anacon
da.org/m-labs/label/main/win-64/pygit2-0.24.0-py35_4.tar.bz2
pygments                  2.1.3                    py35_0    defaults
pyqt5                     5.5.1                    py35_0    http://conda.anacon
da.org/m-labs/label/main/win-64/pyqt5-5.5.1-py35_0.tar.bz2
pyqtgraph                 0.9.10          py_112+git7d34e22    http://conda.anac
onda.org/m-labs/label/main/noarch/pyqtgraph-0.9.10-py_112+git7d34e22.tar.bz2
pyserial                  2.7                      py35_0    defaults
python                    3.5.2                         0    defaults
python-dateutil           2.4.2                     <pip>
python-levenshtein        0.12.0                    <pip>
pythonparser              0.0                        py_6    http://conda.anacon
da.org/m-labs/label/main/noarch/pythonparser-0.0-py_6.tar.bz2
pytz                      2016.4                   py35_0    defaults
qt5                       5.5.1                         1    http://conda.anacon
da.org/m-labs/label/main/win-64/qt5-5.5.1-1.tar.bz2
quamash                   0.5.5                      py_1    http://conda.anacon
da.org/m-labs/label/main/noarch/quamash-0.5.5-py_1.tar.bz2
regex                     2015.11.22               py35_1    http://conda.anacon
da.org/m-labs/label/main/win-64/regex-2015.11.22-py35_1.tar.bz2
scipy                     0.17.1              np111py35_1    defaults
setuptools                23.0.0                   py35_0    defaults
sip                       4.16.9                   py35_2    defaults
six                       1.10.0                   py35_0    defaults
snowballstemmer           1.2.1                    py35_0    defaults
sphinx                    1.4.1                    py35_0    defaults
sphinx-argparse           0.1.13                     py_4    http://conda.anacon
da.org/m-labs/label/main/noarch/sphinx-argparse-0.1.13-py_4.tar.bz2
sphinx-rtd-theme          0.1.9                     <pip>
sphinx_rtd_theme          0.1.9                    py35_0    defaults
vs2015_runtime            14.0.25123                    0    defaults
wheel                     0.29.0                   py35_0    defaults
zlib                      1.2.8                    vc14_3  [vc14]  defaults
```

jordens, 2016-07-12:
#506 from the looks of it.

sbourdeauducq, 2016-07-12:
Sounds like Python 3.5.2 broke TCP/IP connections with asyncio ProactorEventLoop. ``conda install python=3.5.1`` for now.

Maybe one day, ARTIQ will run without monkey patches...

r-srinivas, 2016-07-12:
Downgrading to python 3.5.1 worked. Is the version available through conda enough to test the new contract stuff like editing datasets, RPC pause, seamless handover etc.?

sbourdeauducq, 2016-07-12:
Try it on Linux which doesn't have this bug.

r-srinivas, 2016-07-12:
Okay. The version according to the conda list that was installed was

```
artiq                     2.0.dev         py_460+gitd90fd7d    http://conda.anac
onda.org/m-labs/label/dev/noarch/artiq-2.0.dev-py_460+gitd90fd7d.tar.bz2
```

sbourdeauducq, 2016-07-12:
Note that the monkey patches that allowed ARTIQ to run despite two problems in Python 3.5.1 are not in ARTIQ 2.x anymore. Those two issues are fixed in Python 3.5.2, but they introduced this one. So ARTIQ 2.x cannot be used on Windows for now (unless you reintroduce the monkey patches and downgrade to 3.5.1).

sbourdeauducq, 2016-07-12:
#506

closed 2016-07-12 by sbourdeauducq