m-labs/artiq #509: TTLs pulse when flashing coredevice

r-srinivas, 2016-07-12:
I've noticed the flashing the board pulses all the ttls for a while. This occurs at the beginning and towards the end of the flashing process (the second one appears after the pause after second 79). The first pulse is for about 700 ms, the second for about 3.5 s. It seems to occur for all TTL channels. It's probably not a major issue but should be documented somewhere so people are aware of it.

sbourdeauducq, 2016-07-12:
Reflashing the board necessarily involves swapping bitstreams, with times during which the IO behavior is defined by the hardware only.

jordens, 2016-07-12:
Does this really surprise you? We also don't document that all TTLs go to some state when you turn the device off.

jordens, 2016-07-12:
As a sidenote: what happens (during flashing and when off) depends to a large part on the hardware adapter. I made the proxy bitstreams behave similar to what xilinx does: for pipistrello unused pins are pullups, for kc705 they are floating. During the bitstream loading there might be other values. In general, if I were to design hardware adapters, I would add pullups/downs *and* an interlock circuit that ensures a certain state of the adapter when the device is not operative.

dhslichter, 2016-07-12:
@r-srinivas the simplest way to handle this would be to turn off the 4V supply to the ARTIQ crate while flashing, while leaving the 5V supply on.  This will force all of the TTL outputs to logic 0 at the TTL breakout boards until the 4V supply is turned back on.

closed 2016-07-13 by sbourdeauducq