m-labs/artiq #513: artiq_flash TCL script "Can't find board/pipistrello.cfg"

jboulder, 2016-07-14:
```
((py35dev)) britton@boshi:~/github/m-labs$ conda list artiq
# packages in environment at /home/britton/miniconda3/envs/py35dev:
#
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/channel/main
artiq                     2.0.dev         py_484+gitc50d436    m-labs/label/dev
artiq-kc705-nist_clock    2.0.dev         py_484+gitc50d436    m-labs/label/dev
artiq-pipistrello-nist_qc1 2.0.dev         py_484+gitc50d436    m-labs/label/dev
```
```
((py35dev)) britton@boshi:~/github/m-labs$ 
((py35dev)) britton@boshi:~/github/m-labs$ ls
artiq  asyncserial  migen  misoc
((py35dev)) britton@boshi:~/github/m-labs$ find . -name 'pipistrello*'
./migen/build/lib/migen/build/platforms/pipistrello.py
./migen/migen/build/platforms/pipistrello.py
./artiq/artiq/gateware/targets/pipistrello.py
./misoc/build/lib/misoc/targets/pipistrello.py
./misoc/misoc/targets/pipistrello.py
```
```
((py35dev)) britton@boshi:~/github/m-labs$ 
((py35dev)) britton@boshi:~/github/m-labs$ artiq_flash -t pipistrello -m nist_qc1
Design name: b'top.ncd;UserID=0xFFFFFFFF\x00'
Partname b'6slx45csg324\x00'
Date b'2016/07/13\x00'
Time b'17:34:58\x00'
found binary data length: 1317434
Open On-Chip Debugger 0.10.0-dev-00321-gd4b7cbf (2016-05-26-15:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find board/pipistrello.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/py35dev/bin/artiq_flash", line 9, in <module>
    load_entry_point('artiq==2.0.dev0+484.gc50d436', 'console_scripts', 'artiq_flash')()
  File "/home/britton/miniconda3/envs/py35dev/lib/python3.5/site-packages/artiq/frontend/artiq_flash.py", line 136, in main
    "-c", "; ".join(prog),
  File "/home/britton/miniconda3/envs/py35dev/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', '/home/britton/miniconda3/envs/py35dev/lib/python3.5/site-packages/share/openocd/scripts', '-f', 'board/pipistrello.cfg', '-c', 'init; jtagspi_init 0 {/home/britton/miniconda3/envs/py35dev/lib/python3.5/site-packages/artiq/binaries/pipistrello-nist_qc1/bscan_spi_xc6slx45.bit}; jtagspi_program {/tmp/tmpq4pjwu6u} 0x0; jtagspi_program {/home/britton/miniconda3/envs/py35dev/lib/python3.5/site-packages/artiq/binaries/pipistrello-nist_qc1/bios.bin} 0x170000; jtagspi_program {/home/britton/miniconda3/envs/py35dev/lib/python3.5/site-packages/artiq/binaries/pipistrello-nist_qc1/runtime.fbi} 0x180000; xc6s_program xc6s.tap; exit']' returned non-zero exit status 1

```

sbourdeauducq, 2016-07-14:
How did you install openocd?

jordens, 2016-07-15:
@whitequark i don't see how this would ever have worked. the scripts are not in sitepackages. is this change (a7e101a) ok?

whitequark, 2016-07-15:
@jordens Hm, I tested it only on lab. and seems like openocd also always searches system paths, so that's how I missed it. Change seems OK.

jordens, 2016-07-15:
ack

closed 2016-07-15 by jordens

jboulder, 2016-07-18:
Confirmed OK. Thanks!

```
(artiq20) britton@boshi:~/github/m-labs/artiq/artiq/frontend$ conda list artiq
# packages in environment at /home/britton/miniconda3/envs/artiq20:
#
artiq                     2.0.dev         py_484+gitc50d436    m-labs/label/dev
artiq-pipistrello-nist_qc1 2.0.dev         py_484+gitc50d436    m-labs/label/dev
llvmlite-artiq            0.10.0.dev              py35_24    m-labs/label/main
```