m-labs/artiq #515: Scale/units in datasets

r-srinivas, 2016-07-15:
Would it be possible to have scale/units for datasets? Would help make some of values easier to read.

sbourdeauducq, 2016-07-18:
How do you want this to work at the HDF5 level?

r-srinivas, 2016-07-18:
Would it be possible/make sense to leave it unchanged at the HDF5 level and have the scale just for the GUI?

jordens, 2016-07-18:
How should the browser display it then? And what should happen if the browser is asked to push the dataset from hdf5 to the master?

dleibrandt, 2016-07-18:
I agree that this would be useful.  It seems like at a minimum the scale and units would have to be saved in the dataset database (for persistent datasets).  Perhaps they should also be saved in the hdf5, which would allow the browser to display the units also?

dhslichter, 2016-08-03:
Could one implement this by having datasets take optional units fields?  If populated with a recognized SI unit, display in the dataset list would handle the appropriate prefix notation for the unit (e.g. 10000 with units "Hz" would display as 10 kHz).  In the HDF5, the dataset would have two subfields, one containing the data and another subfield for the unit definition.  

When creating a dataset, one would call `set_dataset(name, units="Hz")` for example to set things up.

sbourdeauducq, 2016-08-04:
> e.g. 10000 with units "Hz" would display as 10 kHz

What about 10001? Or 1000001?

dhslichter, 2016-08-04:
For display purposes in the dataset list, one could also add a field "display_precision" in `set_dataset` that allows you to choose how many digits after the decimal point are displayed (e.g. 10.0 kHz, 10.001 kHz, 10.00100000 kHz).  The 10001 Hz case is handled fine in the HDF5 file anyway because you are storing the full number and not a truncated representation.  If you have a number where you care about more digits than are available in the significand anyway, then that is a larger problem that has to be addressed with extended precision floats, etc.

dhslichter, 2016-08-04:
In general it seems it might make logical sense for datasets to have the same options that arguments can take, since oftentimes arguments will be saved in datasets, and datasets will be used to populate default values of arguments.  The `NumberValue` and `Scannable` arguments already have units/scale/ndecimals parameters, so why not have numerical datasets have these same parameters?  The display of dataset values in the dataset list could then have formatting that is harmonized in a straightforward way with how it is already done in the experiment UIs.