m-labs/artiq #518: Comments for ttls and dds' not displayed in 2.0

r-srinivas, 2016-07-18:
I'm using 2.0dev+497.g9785b39 on windows. I've loaded the dashboard with the same device_db I've been using using the 1.x versions. For example,

```
    "dds0": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 50, "channel": 0},
        "comment": "unassigned"
    },
    "ttl1": {
        "type": "local",
        "module": "artiq.coredevice.ttl",
        "class": "TTLInOut",
        "arguments": {"channel": 1},
        "comment": "bd1"
    },
```

The comments don't display on the dashboard.

jordens, 2016-07-18:
They are displayed. As tooltips.
The reasoning is that the name of the ttl is already descriptive. I.e. if `dds0` is unassigned, you should either not have it at all in your `device_db.pyon` (comment it out) or leave it as its numeric opaque name. Other used dds should be e.g. `dds_bd`.
Alternatively put all the long text that goes beyond the name into the comment. It is displayed as the tooltip.

closed 2016-07-18 by jordens

r-srinivas, 2016-07-18:
Okay, they display on both windows and linux. Thanks for updating the documentation.

r-srinivas, 2016-09-08:
One issue is that it's a lot easier to see the mapping of ttls to functions when we do it like

```
    #  ttls on crate
    "ttl_bd1":      "ttl1",
    "ttl_bd2":      "ttl2",
    "ttl_bdd":      "ttl3",
    "ttl_tickle":   "ttl4",
    "ttl_rd":       "ttl5",
    "ttl_ram1":     "ttl6",
    "ttl_ram2":     "ttl7",
    "ttl_bd_ne":    "ttl8",
    "ttl_ram1_ne":  "ttl9",
    "ttl_ram2_ne":  "ttl10",

    #  pmt/loading box    
    "ttl_pmt":      "ttl20",
    "ttl_flipper":  "ttl21",
    "ttl_oven":     "ttl22",
    "ttl_pi":       "ttl23",

    #  pdqs
    "ttl_pdq_trig": "ttl32",
    "ttl_qc_q1_6":  "ttl33",
    "ttl_qc_q1_7":  "ttl34",
```

It's more difficult to see the mapping when you're describing the whole device as each ttl channel then requires 5-6 lines. I suppose we could just change the name, and map in reverse. It was kind of nice the way it was in 1.x where we could just have the hardware defined generally in the beginning and just map everything in the end, and be able to see all the mappings on a single screen. 

Given how large the box for each ttl channel is anyway, it doesn't save you any space to use the tooltip instead of just having it the way it was where it was "name : comment".