m-labs/artiq #522: Include check_pause in Pauses section of Management System

r-srinivas, 2016-07-19:
It could be useful to have a description of `check_pause` included in the Pauses section,

https://m-labs.hk/artiq/manual-master/management_system.html?highlight=pause

jordens, 2016-07-20:
We do expect that you look around and easily find https://m-labs.hk/artiq/manual-master/management_system.html?highlight=pause#artiq.master.scheduler.Scheduler.check_pause Did you find that?
The API documentation needs to be thorough since it is authoritative and much easier to maintain. Overview and tutorial chapters don't need to be and should not be comprehensive.

r-srinivas, 2016-07-20:
I did see that. It just seemed that this section was unchanged from the release version, though the pause functionality has changed quite a bit with the inclusion of check_pause. I thought it's useful to perhaps just mention that such a function exists in that section so people can be aware of it.

jordens, 2016-07-20:
Ok.

commit ecda9470 (2016-07-21): doc/management_system: point out check_pause() (closes #522)
closed 2016-07-21 by jordens