m-labs/artiq #523: Type inference error

dleibrandt, 2016-07-19:
Running 1.2 on linux.  The following code spits an error.  If either of the print commands in the run method are commented out then the error goes away.

```
from artiq.experiment import *

class TestClass():
    def __init__(self, value):
        self.internal_value = value
        
    @portable
    def value(self):
        return self.internal_value

class TestExperiment(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        
        self.a = TestClass(1)
        self.b = TestClass(2.0)
    
    @kernel
    def run(self):
        print(self.a.value())
        print(self.b.value())
```

whitequark, 2016-07-20:
This is expected behavior. User-defined classes in ARTIQ are not parametric and so all instances of the class must have matching types for every field.

dleibrandt, 2016-07-20:
Oh.  Is this documented anywhere?

whitequark, 2016-07-20:
The documentation section for these features is https://m-labs.hk/artiq/manual-master/compiler.html, and it, unfortunately, omits the restrictions on classes. I will add that.

dleibrandt, 2016-07-20:
Thanks.  That'd be helpful.

commit 3dc01014 (2016-08-03): doc: explain monomorphism of user-defined classes.

Fixes #523.
closed 2016-08-03 by whitequark