m-labs/artiq #524: Browser crashes if maximised when no pyon file exists

r-srinivas, 2016-07-19:
Using artiq 2.0.dev+485.gc0d5914 on windows, if you start the browser for the first time and immediately maximise it, it crashes with the following error message,

```
[artiq-dev-master] c:\artiq-dev-master\artiq-magtrap>python -m artiq.frontend.ar
tiq_browser
INFO:browser:artiq.gui.state:State database 'C:\Users\rabi\AppData\Local\m-labs\
artiq\2\artiq_browser.pyon' not found, using defaults
QWidget::repaint: Recursive repaint detected 
```

If you resize it manually first and then maximise, it's fine. Once the pyon file exists maximising seems to be fine. The issue only comes up if you start it for the first time without the pyon file.

jordens, 2016-07-20:
I can reproduce this. Same version, windows 7, same recipe. Could you try a) reproduce it with a newer version (there were changes related to the log painting)? And then b) could you delete/comment out the `paintEvent()` method (https://github.com/m-labs/artiq/blob/aa1f32a5df4dcf76ad271e804081b683c49895e9/artiq/browser/experiments.py#L419) and see whether this changes anything?

r-srinivas, 2016-07-20:
I used `2.0.dev py_503+git2099344` and I still see this issue. Commentin out the `paintEvent()` method didn't fix it.

jordens, 2016-07-20:
Thanks. I'll try to reproduce but it doesn't smell like a deeper Qt bug. Any chance you could get a stack trace using ms visual c++?
@whitequark can you help or do you see something to test?

jordens, 2016-07-20:
Maybe https://bugreports.qt.io/browse/QTBUG-48823
Could you try your recipe with `python -m artiq.frontend.artiq_browser --browse-root c:\artiq-dev-master\artiq-magtrap c:\artiq-dev-master\artiq-magtrap`?

r-srinivas, 2016-07-20:
That seems to maximise without any issues.

jordens, 2016-07-20:
Ok. Sounds like funky icons and that Qt bug then.
@whitequark @sbourdeauducq do we plan to land qt 5.6.1?

jordens, 2016-07-23:
depends on #261 (in all likelihood)
we could try finding a work around but since this usually only happens once there would be little benefit.

sbourdeauducq, 2017-06-26:
Fixed.

closed 2017-06-26 by sbourdeauducq