m-labs/artiq #525: memory failure on nist_clock gateware

jmizrahi, 2016-07-21:
When I flash the nist_clock gateware to the kc705, I get a memory initialization failure. This does not happen when I flash the nist_qc1 gateware. Here are the details:

I built the clock gateware using:
`$ python3.5 -m artiq.gateware.targets.kc705 -H clock`

Then I flashed from a directory with the binaries using:
`$ artiq_flash -d . -t kc705`

I then looked at startup using flterm, and got:

```
MiSoC BIOS
(c) Copyright 2007-2016 M-Labs Limited
Built Jul 15 2016 16:39:35

BIOS CRC passed (145cef98)
Initializing SDRAM...
Write leveling: 15* 16* 13  13   8   7   3   6  completed
Read bitslip: 7 6 5 4 3 
Read delays: 7:01-12  6:00-12  5:03-13  4:32-33  3:10-21  2:00-06  1:00-09  0:00-09  completed
Memtest failed: 256512/532736 words incorrect
Memory initialization failed
BIOS> 
```

This is somewhat intermittent -- occasionally it would pass memtest, and I couldn't figure out the pattern. But it would fail much more often than it would succeed. Also, not surprisingly, I am unable to communicate with the kc705 over ethernet.

I tried switching to the release-1 branch, re-built the gateware and got the same results.

When I repeated all of the above with the qc1 gateware, these problems go away -- the memory test is fine and the ethernet communication works.

sbourdeauducq, 2016-07-22:
Occasionally, ISE miscompiles and this results in problems like that. Vivado seems less prone to this sort of bug; add ``--toolchain vivado`` to your bitstream build command.

jmizrahi, 2016-07-22:
That worked, thanks!

Perhaps Vivado should be the default toolchain?

sbourdeauducq, 2016-07-22:
Not sure; ISE can compile for both Spartan-6 and 7-series while Vivado only supports 7-series. Maybe when S6 becomes more obsolete.

closed 2016-07-22 by sbourdeauducq

jbqubit, 2016-07-22:
>
> Occasionally, ISE miscompiles and this results in problems like that.
> Vivado seems less prone to this sort of bug; add --toolchain vivado to
> your bitstream build command.
​Please update documentation to read something like "If targeting Xilinx
devices other that Spartan6 add --toolchain vivado​ to bitstream build
command."

sbourdeauducq, 2016-07-22:
That's Xilinx's problem, not ours.

sbourdeauducq, 2016-07-22:
And Vivado also miscompiles from time to time, just less frequently than ISE in my experience.

jbqubit, 2016-07-22:
> That's Xilinx's problem, not ours.
>
​I'm not suggesting we fix bugs in the Xilinx tool chain. Rather, tell
users how to configure the build system to maximize success. ​

jordens, 2016-07-22:
Joe (is this another of your accounts?), sprinkling the documentation with ad-hoc and out-of-context notes is shortsighted because nobody does the necessary job of regularly verifying that the information is still the best advice and worse, eventually nobody will remember why it was added in the first place. There is often a much better technical solution than asking for something to be added to the documentation and hoping to be done with it.
We might choose a better default in this case.

jbqubit, 2016-07-22:
"There is often a much better technical solution than asking for something
to be added to the documentation and hoping to be done with it."

A compiler flag is a technical solution to ensure that the correct compiler
version is used. ARTIQ already uses meta.yaml to ensure that build
dependencies are versioned correctly. Do you recommend a different
technical solution? -Joe

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

sbourdeauducq, 2016-07-23:
Everyone who has done medium-sized FPGA projects had to deal with Xilinx toolchain bugs like that. The ARTIQ documentation should not be about writing gateware.

If we change the default, we should do that in MiSoC and not ARTIQ.

jboulder, 2016-07-23:
>
> >If we change the default, we should do that in MiSoC and not ARTIQ.
>
That's a technical solution that would prevent recurrence of the specific
error that Jonathan encountered and satisfies Robert's desire to not modify
the documentation. And it reflects the evolution of Xilinx build tools to
Vivado. -Joe