m-labs/artiq #526: Compiler optimization

dleibrandt, 2016-07-21:
Running 1.2 on linux.  The following code spits an error, I assume because string concatenation is not supported on the core device.  It would be convenient if the string concatenation were precomputed by the compiler, at least for this case where self.name is a kernel invariant.  Would it be possible to make this happen?

```
import numpy as np
from artiq.experiment import *

class CompilerOptimization(EnvExperiment):
    kernel_invariants = {"name"}
    
    def build(self):
        self.setattr_device("core")
        self.name = "CompilerOptimization"
    
    def prepare(self):
        self.set_dataset(self.name + ".dataset", np.array([0]*5), broadcast=True)
        
    @kernel
    def run(self):
        self.mutate_dataset(self.name + ".dataset", 0, 1)
```

whitequark, 2016-07-21:
Being marked as kernel invariant doesn't mean that `self.name` is in every case known at compile time, it just means that it doesn't change and so it can be hoisted out of loops, propagated across the call graph, etc.

There is no inherent problem in string concatenation in kernel code though; this is probably the only string operation that can be easily and safely done there. So that can be simply implemented.

dleibrandt, 2016-07-21:
Oh OK.  I guess I was misunderstanding what kernel invariants are.  If it's easy to add string concatenation to the kernel, that would be great!  Alternatively, I don't know if it would be generically useful to have a way to declare variables as constants in artiq python (I realize this isn't possible in regular python).

whitequark, 2016-07-21:
> Alternatively, I don't know if it would be generically useful to have a way to declare variables as constants in artiq python (I realize this isn't possible in regular python).

See #193. There is no good syntax for this feature (that still allows kernels to be written in regular Python), and in general any implementation compatible with regular Python is bound to be fragile.

r-srinivas, 2016-07-22:
I've been doing this and haven't run into any issues,

```python
 def prepare(self):
        self.xvar = "data.z_shim"
        self.xlabel = self.xvar + ".x"
        self.counts_label = self.xvar + ".counts"
        self.counts_hist_label = self.xvar + ".counts_hist"
        self.counts_hist_bins_label = self.xvar + ".counts_hist_bins"
        l = len(self.z_shim_scan)
        #  set up size of arrays to hold data
        self.set_dataset(self.xlabel, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_label, np.full(l, np.nan),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_label, np.empty((l, int(self.repetitions))),
                         broadcast=True, save=True)
        self.set_dataset(self.counts_hist_bins_label, np.linspace(0, 100, self.nbins+1),
                         broadcast=True, save=False)
        self.counts = [0]*int(self.repetitions)
        #  generate list of scan variable as numpy arrays not supported
        #  on core device
        self.z_shims = list(self.z_shim_scan)

def run(self):
.
.
.
        self.mutate_dataset(self.counts_hist_label, index, hist)
        self.mutate_dataset(self.counts_label, index, total/self.repetitions)
        self.mutate_dataset(self.xlabel, index, y_shim)
```

commit 1a518ea7 (2016-08-08): compiler.embedding: implement string concatenation.

Fixes #526.
closed 2016-08-08 by whitequark