m-labs/artiq #530: Increasing ofifo and ififo depths

r-srinivas, 2016-07-28:
We've been having some issues with underflow errors on pulse sequences, and one solution seems to be to increase the ofifo depth here,

```python
class NIST_QC2(_NIST_Ions):
    """
    NIST QC2 hardware, as used in Quantum I and Quantum II, with new backplane
    and 24 DDS channels.  Two backplanes are used.  
    """
    def __init__(self, cpu_type="or1k", **kwargs):
        _NIST_Ions.__init__(self, cpu_type, **kwargs)

        platform = self.platform
        platform.add_extension(nist_qc2.fmc_adapter_io)

        rtio_channels = []
        clock_generators = []

        # All TTL channels are In+Out capable
        for i in range(40):
            phy = ttl_serdes_7series.Inout_8X(
                platform.request("ttl", i))
            self.submodules += phy
            rtio_channels.append(rtio.Channel.from_phy(phy, ififo_depth=512))
```

We've tried changing ofifo_depth to 1024 and haven't had any issues yet. Is there a limit to how deep it can be? Would it be possible to have ofifo_depth=1024 and ififo_depth=128? Are there any potential problems to increasing this?

jordens, 2016-07-28:
The only limit to fifo depth is the amount of available block ram. You can check your resource usage in the vivado outputs. I don't see any immediate technical limits other than that one.

A fifo deeper than -- let's say -- your estimate of the number of events on any one channel in an experiment won't be a solution, whether it seems to be one or not. It should be very clear that playing with the fifo size (or manually converting for machine units in a blanket fashion) to resolve underflows for "normal" experiments is only delaying exposing the actual problem here.

The better estimate for a good fifo size is the maximum burst size you expect on a channel. A burst is the maximum number of events that you expect to have to emit at a rate faster than the CPU. That would be pretty small (< 100) for normal experiments and maybe (just maybe) larger for benchmarking stuff. But benchmarking stuff should not be a burst above the CPU capacity because that can't scale.

Please try to identify the actual problem here that leads to the underflow. I recommend spending zero time on manually converting to machine units or finding the absolute maximum of fifo depth. A logic analyzer trace would help. And a minimal example experiment.

r-srinivas, 2016-07-28:
The experiment we were having trouble with is the following,

```python

class Cooling_Axial_Frequency_Scan(EnvExperiment):
    '''Cooling Axial Frequency Scan'''

    def build(self):
        #  build hardware for experiment
        self.setattr_device("core")
        self.setattr_device("ttl_bdd")
        self.setattr_device("ttl_ram1")
        self.setattr_device("ttl_ram2")
        self.setattr_device("ttl_ram1_ne")
        self.setattr_device("ttl_ram2_ne")
        self.setattr_device("core_dds")
        self.setattr_device("dds_ram1")
        self.setattr_device("dds_ram2")
        #  variable to be scanned
        self.d0 = self.get_dataset("raman.light_shift")
        self.a0 = self.get_dataset("raman.axial.f")        
        self.f0 = (self.get_dataset("mw.f.3p3_2p2") +
                   self.get_dataset("mw.d.3p3_2p2") +
                   self.d0)/4
    
        self.setattr_argument("axial_f",
                              Scannable(
                              default=LinearScan(self.a0-0.2e6,
                                                 self.a0+0.2e6,100),
                              scale=1e6, unit='MHz', ndecimals=3), "Scan")
        self.setattr_argument("t_pulse", NumberValue(6e-6, scale=1e-6,
                                                     unit="us"), "Scan")
        #  instance of 1d scan subcomponent    
        self.scan = Scan_1D(parent=self, xvar="data.cooling.axial.f")
        #  instance of detection subcomponent
        self.detect = Detection(parent=self)
        #  instance of cooling subcomponent
        self.cool = Cooling(parent=self)
        #  call fitting function
        self.fit = Rabi_Flopping(parent=self, t_pi0=6e-6)    
        #  call value for dds from dataset database
        
#==============================================================================
#  prepare stage for initialising datasets
#==============================================================================

    def prepare(self):
        l = len(self.axial_f)
        self.scan.initialise_datasets(l)
        #  generate list of scan variable as numpy arrays not supported
        #  on core device
        self.frequencies = list(self.axial_f)

#==============================================================================
#  run
#==============================================================================    

    def run(self):
        self.detect.set_det_freq()
        self.experiment(self.frequencies)
        
    #  using this function to have all scan points on kernel
    @kernel
    def experiment(self, frequencies):
        rtio_log("dds_ram1")
        i = 0
        for f in frequencies:
            self.scan_point(i, f)
            rtio_log("ttl_ram1", "i", i)
            i += 1
[... 110 lines cut ...]
        for i in range(30):
            self.pulse_raman(time)
            self.rd_repump(1600)
#            delay(1*us)        
            
    @kernel
    def pulse_raman(self, time):
        with parallel:
            self.ttl_ram1_ne.pulse_mu(time)
            self.ttl_ram2_ne.pulse_mu(time)
            self.ttl_ram1.pulse_mu(time)
            self.ttl_ram2.pulse_mu(time)
```

The issue we had was looking at the RTIO analyzer, we were losing a lot of slack per raman cool pulse, mainly because of the microwave pulses we have to carry out for the repump. Without using machine units, a for loop with just one loop for

```python
    @kernel
    def raman_cool(self, time):
        for i in range(30):
            self.pulse_raman(time)
            self.rd_repump(1600)
#            delay(1*us)        
```
Would work, but even two loops would result in an underflow. This slack was reduced drastically by converting to machine units, and with the additional fifo depth, we are able to 30 pulses with the slack going from about to `500us` to '200us` at the end of the sequence of 30 pulses. I'll try to get the traces we had from yesterday again.

dhslichter, 2016-07-29:
@jordens is correct that increasing ofifo or ififo depths does not scale arbitrarily.  However, it seems reasonable to make the queues deeper if the resources are available on the FPGA, as this reduces the chance of underflow errors for complex pulse sequences with many pulses of short duration.  We will submit a patch to the nist_qc2 gateware increasing FIFO depths to the depth that seems necessary for our experiments (~several hundred pulses for sideband cooling from high nbar).  If we want to do long randomized benchmarking with very fast microwave pulses, we might need deeper FIFOs and can make a custom bitstream ourselves for the necessary channel(s).  

The specific case above that @r-srinivas presents shows that having deeper FIFOs and converting to machine units can be helpful in a specific case with lots of short duration pulses, that's all.  Since these are easy and straightforward things to do, it's not an issue.

dhslichter, 2016-07-29:
I think we can safely close this one.

jordens, 2016-07-29:
Sure. But I still suspect that you would be better off with a solution that does not require either.

closed 2016-07-29 by jordens