m-labs/artiq #531: Controller methods should automatically be RPCs

dleibrandt, 2016-07-28:
I'm using the novatech controller in an experiment.  If I try to call the novatech methods from the core device as below, I get an AssertionError.
```
from artiq.experiment import *

class NovatechTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.novatech = self.get_device("novatech")
    
    @kernel
    def run(self):
        self.novatech.set_freq(1, 100.)
```

If I wrap the novatech methods such that they are RPCs as below, I don't get an error.
```
from artiq.experiment import *

class NovatechTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.novatech = self.get_device("novatech")
    
    def set_novatech(self):
        self.novatech.set_freq(1, 100.)
    
    @kernel
    def run(self):
        self.set_novatech()
```

I think it would be more convenient if all controller methods were RPCs by default.

whitequark, 2016-07-28:
You shouldn't be ever getting an assertion failure, that means a bug regardless of anything else (like whether the behavior is permitted). What is the backtrace? (Alternatively, what's your device_db?)

dleibrandt, 2016-07-28:
Backtrace:
```
ERROR:worker(2430,NovatechTest.py):root:Terminating with exception (AssertionError)
Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 455, in _unify_attribute
    attributes, attr_value_type = self.attr_type_cache[attr_type_key]
KeyError: (140341427382312, 'novatech')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 226, in main
    exp_inst.run()
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/coredevice/core.py", line 105, in run
    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs, set_result)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/coredevice/core.py", line 86, in compile
    stitcher.finalize()
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 547, in finalize
    inferencer.visit(self.typedtree)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 1222, in visit_FunctionDefT
    self.generic_visit(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 34, in _visit_one
    return self.generic_visit(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 859, in visit_CallT
    self.generic_visit(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 90, in visit_AttributeT
    self.generic_visit(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 93, in visit_AttributeT
    loc=node.loc)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 458, in _unify_attribute
    self._compute_value_type(object_value, object_type, object_loc, attr_name, loc)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 436, in _compute_value_type
    ast = self.quote(attr_value, object_loc.expanded_from)
  File "/home/rabi/anaconda3/envs/artiq-1.2/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 844, in _quote
    node = synthesizer.quote(value)
[... cut ...] /python3.5/site-packages/artiq/compiler/embedding.py", line 186, in quote
    assert isinstance(value.kernel_invariants, set)
AssertionError
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 2430
```

device_db:
```
    "novatech": {
        "type": "controller",
        "best_effort": False,
        "host": "::1",
        "port": 3269,
        "target_name": "novatech409b",
        "command": "novatech_controller --bind {bind} -p {port} --simulation"
    },
```

sbourdeauducq, 2016-07-29:
@whitequark I suppose your compiler doesn't like the custom ``__getattr__`` of RPC clients.

commit 8a40871f (2016-08-03): protocols.pc_rpc: exclude kernel_invariants from proxying.

Fixes #531.
commit bdb85c8e (2016-08-03): protocols.pc_rpc: exclude kernel_invariants from proxying.

Fixes #531.
closed 2016-08-03 by whitequark