m-labs/artiq #532: "Repeat single value" option in `scannable`

dtcallcock, 2016-07-29:
When we are tweaking something manually on the experiment (eg. a mirror mount) we often want to continuously repeat a single experiment, without changing anything else, to see the effect of the tweaking.  In hfGUI this was called a 'dummy' scan.  Currently we do this in artiq by doing a linear scan but making the range so small there is effectively no change in the parameter.  Whilst this generally works fine it is a bit annoying to set up and I think we do this often enough in the lab to merit a single-click implementation.  A couple of possible implementations might be:

- Add a new type of `ScanObject` to `scannable` for 'repeat single value'.  This would have 'value' and 'number of repetitions' arguments.

- Add a 'repetitions' argument to the existing 'No scan' `ScanObject`.  The problem with this is that when you go to scan another variable you will have to remember to set this back to '1' to avoid doing a 2D scan.

jordens, 2016-07-29:
+1 for the latter

sbourdeauducq, 2016-07-30:
Agreed.

The "disable other scans" button (in 2.x) can also set the repeat count to one.

jordens, 2016-08-15:
You can always add as many dummy variables as you want. How you handle them is your choice. But that's a use case and not a distinct feature, right?
If you want to plot against indices, just generate an indices dataset.
If you want the applet to magically ignore the abscissa values and use indices, you might have to change the applet code.

sbourdeauducq, 2016-08-16:
The experiment can also detect when this feature is used (the code would be something like ``isinstance(scan, NoScan) and scan.repeat > 1``) and generate the corresponding datasets for applet consumption.

sbourdeauducq, 2016-09-08:
@r-srinivas 7efd9905412cd96db9744abcce06c1233b3b43df was necessary to avoid unnecessary typing and keep backwards compatibility. You could have found the problem by running (or just scanning) the arguments_demo example.

closed 2016-09-08 by sbourdeauducq

r-srinivas, 2016-09-08:
Okay. I had actually thought about it but the other scan options didn't have default values, but I suppose they don't have arguments like repetitions that are not as necessary.

r-srinivas, 2016-09-09:
Could this be added to 2.0 or would it be considered a new feature that you don't want to introduce in a version?

jordens, 2016-09-09:
i am ok with making an exception

dhslichter, 2016-09-12:
@sbourdeauducq, thoughts?  This would be very nice to have in 2.0, and AFAICT shouldn't break back compatibility with existing code.

sbourdeauducq, 2016-09-13:
Ok.

commit 9b937e08 (2016-09-08): added repetitions for no scan, repetitions set to one when disable other scans selected. Closes #532
commit d4af4414 (2016-09-13): added repetitions for no scan, repetitions set to one when disable other scans selected. Closes #532