m-labs/artiq #533: Digits after decimal rejected in scan widget

dhslichter, 2016-08-01:
It appears that the ghost of #457 may still be with us -- in artiq 1.2 py_0+gitce73e8e on Windows, when typing values into the high or low limits of a `ScanWidget`, it usually (but not 100% of the time) refuses to accept values after the decimal point is typed.  For example, if I want to enter 2.53 MHz, I delete the existing value (field is then empty except for MHz unit string), then type 2.53.  Sometimes it gives me 2.000 MHz, sometimes it will let me get the 5 in for 2.500 MHz, but I can never get the 3 for 2.530.  Digits to the right of the decimal place have to be added one at a time by deleting a digit and typing in a new one in its place.  

```        self.setattr_argument("blue_axial_f",
                              Scannable(
                              default=LinearScan(self.a0-0.2e6,
                                                 self.a0+0.2e6,100),
                              scale=1e6, unit='MHz', ndecimals=3), "Scan")      
```

jordens, 2016-08-01:
Pretty certain that this is mostly orthogonal to #457.
A quick check on a suspicion I have. Does your success in reproducing it depend on whether you would be narrowing down or widening the range by the number you want to enter?

dhslichter, 2016-08-01:
There is some behavior along these lines, now that I check in more detail.  If I always start by setting the lower limit to 5.000 MHz and the upper limit to 7.000 MHz, I get the following: changing the lower limit to 5.X or the upper limit to 7.X (i.e., highlighting and replacing the whole number but using the same digit before the decimal point as previously) always seems to work for any single digit X, but never for two digits after the decimal place.  If I start with 5.X or 7.X, I never get any digits after the decimal place, no matter what.  If I try to change the number before the decimal place (e.g. going to 8.X, 4.X, or 6.X), I never get any digits after the decimal place.  If my upper limit starts out with more than one digit to the left of the decimal (e.g. 12.000 MHz), I never get any digits to appear after the decimal place no matter what.

dhslichter, 2016-08-01:
The upshot here is that it doesn't really appear to be about widening or narrowing the range per se, but there are some patterns.

commit 14addf74 (2016-08-04): gui.entries: avoid intermediate value feedback, closes #533
commit f183f878 (2016-08-04): gui.entries: avoid intermediate value feedback, closes #533
closed 2016-08-04 by jordens