m-labs/artiq #535: hard FP support on the kernel CPU

sbourdeauducq, 2016-08-05:
https://github.com/m-labs/artiq/issues/529#issuecomment-237135917

dhslichter, 2016-08-05:
Awesome, thanks for adding this.  Can you post this on the m-labs website in the ARTIQ extensions list as well?

sbourdeauducq, 2016-08-20:
I intend to remove this page at some point and replace it with a link to https://github.com/m-labs/artiq/issues?utf8=%E2%9C%93&q=label%3Atype%3Afor-contract%20

sbourdeauducq, 2017-03-18:
@dhslichter mentioned basic math functions (sin, cos, log, exp). Should those be gateware-accelerated, or are slower software implementations (using gateware FP for floating addition, multiplication, etc.) OK?

jordens, 2017-03-18:
I would really prefer not to stray to far of the beaten path here, i.e. for hard FP and vectorization (which is an extremely complicated thing to reinvent) we should concentrate on the corresponding openrisc instruction sets and implement them, instead of inventing new instruction sets. Also accelerated trigonometry is of similar complexity as vectorization.
Since hard FP and trigonometry have a multiplicative impact on performance, focusing on "standard" hard FP first seems prudent.

dhslichter, 2017-03-20:
I agree with @jordens; the most important thing to do here would be to implement a "standard" hard FP instruction set.  Basically the thing to do would be to implement all the instructions defined in the openrisc standard (ORFPX32) but not actually implemented in mor1kx.  The sin/cos/log/exp functions could then be constructed from these with (hopefully) reasonable performance, and we can decide later if they truly require hardware acceleration themselves.

jordens, 2017-05-11:
@whitequark could you add a short assessment of the state/quality/existence of support for or1k FP in the toolchain (artiq compiler, llvm, llvmlite/new-pyllvm, runtime) and the biggest roadblocks to this issue?

whitequark, 2017-05-11:
> artiq compiler
> llvmlite/new-pyllvm
> artiq runtime
> testbenches

No changes required as the LLVM instructions are the same for soft or hard FP.

> compiler-rt

No changes needed as we will be using *less* compiler-rt code afterwards.

> llvm-or1k

The OR1K backend supports the complete OR1K FPU single-precision instruction set except the MAC instruction. Extending to double precision (which we use) is trivial. Ditto for MAC.

jordens, 2017-05-11:
@sbourdeauducq would we use the mor1kx FPU or port some other FPU or write ourselves a new one or go some completely different route?

sbourdeauducq, 2017-05-11:
Take the usable parts of the mor1kx FPU (e.g. its interface into the CPU), rewrite the others.

sbourdeauducq, 2021-02-25:
Seems VexriscV has a FPU now and I would expect high quality.

sbourdeauducq, 2021-09-12:
Funded by Duke.

closed 2021-11-08 by sbourdeauducq

sbourdeauducq, 2021-11-08:
speedup: https://github.com/m-labs/artiq/pull/1777#issuecomment-962926054

PR #1777 (2021-11-08): Add floating-point instruction support to kernel for non-Kasli-v1.x targets
# ARTIQ Pull Request

## Description of Changes

### CPU
All targets except Kasli v1.x:
- Enabled an FPU from VexRiscv in kernel_cpu (see VexRiscv_G variant of VexRiscv in MiSoC)
- Select 64-bits data-path / comm CPU data width, to accommodate the FPU.
- Implement 64-bits support to submodules (DMA, DRTIO).

FPU is not supported in Kasli, due to timing issue.

### Test
Performance is tested by running `mandelbrot.py` on kc705 nist_clock variant.
Before this patch: 6.03563 seconds
After this patch: 3.88476 seconds

DRTI [... cut]