m-labs/artiq #537: No rule to make target 'core/mem.o', needed by 'liblwip.a'.

jboulder, 2016-08-07:
```bash 
$ python3.5 -m artiq.gateware.targets.kc705

(artiq20) britton@boshi:~/github/m-labs/artiq$ python3.5 -m artiq.gateware.targets.kc705

^[[Amake: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libcompiler_rt'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libcompiler_rt'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libbase'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libbase'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libnet'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libnet'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/bios'
 CC       main.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m misoc.tools.mkmscimg bios.bin
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/bios'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/liballoc'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/liballoc'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libm'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libm'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libdyld'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libdyld'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libunwind'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/libunwind'
make: Entering directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/liblwip'
rm -f lwipopts.h
rm -f arch
ln -s /home/britton/github/m-labs/artiq/artiq/runtime/liblwip/lwipopts.h lwipopts.h
ln -s /home/britton/github/m-labs/artiq/artiq/runtime/liblwip/arch arch
mkdir -p core/ipv4
mkdir -p netif
mkdir -p netif/ppp
make: *** No rule to make target 'core/mem.o', needed by 'liblwip.a'.  Stop.
make: Leaving directory '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/liblwip'
Traceback (most recent call last):
  File "/home/britton/miniconda3/envs/artiq20/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/britton/miniconda3/envs/artiq20/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/britton/github/m-labs/artiq/artiq/gateware/targets/kc705.py", line 418, in <module>
    main()
  File "/home/britton/github/m-labs/artiq/artiq/gateware/targets/kc705.py", line 414, in main
    build_artiq_soc(soc, builder_argdict(args))
  File "/home/britton/github/m-labs/artiq/artiq/gateware/soc.py", line 61, in build_artiq_soc
    builder.build()
  File "/home/britton/github/m-labs/misoc/misoc/integration/builder.py", line 149, in build
    self._generate_software()
  File "/home/britton/github/m-labs/misoc/misoc/integration/builder.py", line 126, in _generate_software
    subprocess.check_call(cmd)
  File "/home/britton/miniconda3/envs/artiq20/lib/python3.5/subprocess.py", line 581, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/britton/github/m-labs/artiq/misoc_nist_clock_kc705/software/liblwip']' returned non-zero exit status 2
```

Same error when building for pipistrello.

jordens, 2016-08-07:
`git submodule init`/`git submodule update`

jboulder, 2016-08-07:
Thanks! Wasn't even a bug. :) Resolved. 

```bash
britton@boshi:~/github/m-labs/artiq$ git submodule init
Submodule 'artiq/runtime/lwip' (https://github.com/m-labs/lwip) registered for path 'artiq/runtime/lwip'
britton@boshi:~/github/m-labs/artiq$ git submodule update
Cloning into 'artiq/runtime/lwip'...
remote: Counting objects: 32097, done.
remote: Total 32097 (delta 0), reused 0 (delta 0), pack-reused 32097
Receiving objects: 100% (32097/32097), 6.09 MiB | 7.44 MiB/s, done.
Resolving deltas: 100% (24141/24141), done.
Checking connectivity... done.
Submodule path 'artiq/runtime/lwip': checked out '33955c636d17bee230cde746a92e1cff11a7a177'
```

```bash
$ python3.5 setup.py develop --user
```

```bash
$ python3.5 -m artiq.gateware.targets.kc705
```
Building....

closed 2016-08-07 by jboulder